rsasaki0109/lidar_slam_ros2 #18: cannot load .bag file

CAKGOD, 2020-07-31:
**Required Info:**

- Operating System:
  - ubuntu 20.04
- ROS2 Version:
  - ROS2 foxy

#### Steps to reproduce issue
(1) I build this framework following the[ tutorial](https://github.com/rsasaki0109/lidarslam_ros2#requirement-to-build) in the README.md.
(2) I tried running the [demo](https://github.com/rsasaki0109/lidarslam_ros2#trial-environment) and the problem occured.

#### Problem Description
The input [dataset ](https://github.com/rsasaki0109/lidarslam_ros2#trial-environment)provided by the demo is in the .bag file format, but ros2 only provides access to the .sqlite3 file format. Even if the rosbag2 plugin is installed, the .sqlite file still cannot be read. Framework testing cannot continue.
I am very interested in how you read the .bag file in ros2. Can you provide the system version and ros version, or other solutions? Thanks~

rsasaki0109, 2020-07-31:
To play a rosbag from ros1 on ros2, you need to install ros1.

Next, you need to install rosbag2_bag_v2.
https://github.com/ros2/rosbag2_bag_v2

```
sudo apt install -y ros-foxy-rosbag2-bag-v2-plugins
```
See README.md in rosbag2_bag_v2 for details.

If you have any questions, please comment again.

CAKGOD, 2020-08-02:
@rsasaki0109 Thank you very much for your help. I finally solved the problem. Both ros1 and ros2 need to be installed. In addition, the ROS 1 installation must be sourced before sourcing the ROS 2 installation to make the plugin work. The $ros2 bag info -s rosbag_v2 <path_to_bagfile> and the $ros2 bag play -s rosbag_v2 <path_to_bagfile> can be used. But the $ros2 bag record -s rosbag_v2 <topic name> does not work. Thanks again~

closed 2020-08-02 by CAKGOD

vico1999-ros, 2023-11-09:
Is there any bag can be played on ros2 directly.  both versions of ROS installed on same computer is seem not wise.  finded another data source is more easier

rsasaki0109, 2023-11-15:
@vico1999-ros 
You can use [rosbags](https://pypi.org/project/rosbags/) to convert ROS1 rosbags to ROS2 rosbags without installing ROS1. I will update the readme.md later.

Er-Ding, 2024-07-09:
After I convert .rosbag to .db3 format file and then run the program, I found that the topic /velodyne_points always has message lost problem.How can I solve this problem?
Ubuntu version: 24.04
ROS version: Jazzy

rsasaki0109, 2024-07-09:
You should ask the creator of rosbags about that.
https://gitlab.com/ternaris/rosbags

rsasaki0109, 2024-07-09:
How does the message look when it is converted?

Er-Ding, 2024-07-09:
> Sent Time:2024/7/9 17:09
>
>
> Cc recipient:"Er-Ding"< ***@***.*** &gt;;"Comment"< ***@***.*** &gt;;

When I converted it, the information like fig1.Although the "bag size" showed 0B, but when I check this file's propertise,it's 2.4G
When I play this data and run "ros2 topic echo /velodyne_points ",the result like fig2

fig1:
fig2:

朱军
***@***.***

                       
Original Email

rsasaki0109, 2024-07-09:
The image is not displayed.

Er-Ding, 2024-07-09:
> The image is not displayed.
fig1
[image]
fig2:
[image]

rsasaki0109, 2024-07-09:
The bag size is 0B, so it seems like the conversion didn't go well. It might be worth trying to convert it again.