rsasaki0109/lidar_slam_ros2 #19: __init__() missing 1 required keyword-only argument: 'node_executable'

CAKGOD, 2020-08-09:
Excuse me. I build the lidarslam_ros2 done. A problem occured when i run the command $ ros2 launch lidarslam lidarslam.launch.py. The information is as followed.

[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught exception when trying to load file of format [py]: __init__() missing 1 required keyword-only argument: 'node_executable'

System: ubuntu 18.04
ros version: ros1 melodic + ros2 eloquent

How can i solve this problem? Thank u~

rsasaki0109, 2020-08-09:
Perhaps changing the "executable" part of the lidarslam.launch.p file to "node_executable" will fix the problem?

CAKGOD, 2020-08-09:
thank u ~  i will make a try.

CAKGOD, 2020-08-10:
It works for me. Thanks! And i will close this issue.

closed 2020-08-10 by CAKGOD