rsasaki0109/lidar_slam_ros2 #20: Will the test require modifying the Lidar parameters?

toasttofreedom, 2020-08-21:
Hi,
I found each yaml file has no Lidar parameters. Will the test require modifying the Lidar parameters?
E.g. Can I run this using kitti dataset(Velodyne HDL-64E 3D laser scanner) without modifying any parameters?
Looking forward to your reply!

rsasaki0109, 2020-08-21:
There is no need to change the lidar parameters (VLP-16, HDL-32, etc.) in this package.

rsasaki0109, 2020-08-21:
This package does not have the lidar parameter (VLP-16, HDL-32, etc.).
I think you can run it in kitti without changing any parameters

If it doesn't work, try changing the following parameters.
・Change the vg_size_for_input
・When registration_method is NDT, change ndt_resolution, ndt_num_threads,.
・When registration_method is GICP, change the gicp_corr_dist_threshold.

toasttofreedom, 2020-08-21:
copy that. thanks for your reply.
I will try to run this and I will seek advice from you when I meet problem.

rsasaki0109, 2022-04-20:
I am closing this issue because there is no recent activity on this issue.
Please reopen the issue if you encounter any problems.

closed 2022-04-20 by rsasaki0109