rsasaki0109/lidar_slam_ros2 #21: IMU alignment unclear

dHofmeister, 2020-09-04:
Dear @rsasaki0109,

Should I follow the IMU guide given in: [https://github.com/TixiaoShan/LIO-SAM#prepare-imu-data](https://github.com/TixiaoShan/LIO-SAM#prepare-imu-data)?
I ask this since I assume I dont need to prepare the lidar data as such: [https://github.com/TixiaoShan/LIO-SAM#prepare-lidar-data](https://github.com/TixiaoShan/LIO-SAM#prepare-lidar-data) as pointCloud2 data does not carry ring numbers.

I assume that I can simply make sure myself that the axis of the IMU data align with the axis of the lidar? Or should the output of the IMU be identical to the axis of the IMU image? With all the axis modifications as mentioned?

Does the imu data have to contain RAW acceleration data? (incl. gravity) and do i need to compute a covariance matrix? 

Last question: How can it be that the modules start running when I specify use_imu: True at moments when i am not launching the IMU? Shouldnt it wait for imu messages to come in before starting?  

Im not able to get reliable performance yet. It jumps all over the place. But am hopeful.

Greetings and thank you for your time,

Deniz

rsasaki0109, 2020-09-04:
For IMU composites in this package, the lidar data does not need to include ring numbers or time, as lio-sam does.

Currently, the IMU composite for this package is based on Lego-LOAM. 

Distortion correction by imu
lidar_undistortion.hpp
https://github.com/rsasaki0109/lidarslam_ros2/blob/foxy/scanmatcher/include/scanmatcher/lidar_undistortion.hpp

If you have trouble getting it to work, I recommend looking at the lego-loam issue.

Examples of lego-loam issues
How to align the IMU to Lidar? && Problems with the KITTI dataset 
https://github.com/RobustFieldAutonomyLab/LeGO-LOAM/issues/153
Coordinate transformation when use imu 
https://github.com/RobustFieldAutonomyLab/LeGO-LOAM/issues/19

> I assume that I can simply make sure myself that the axis of the IMU data align with the axis of the lidar? Or should the output of the IMU be identical to the axis of the IMU image? With all the axis modifications as mentioned?

I think the above issues are helpful.

> Does the imu data have to contain RAW acceleration data? (incl. gravity) and do i need to compute a covariance matrix?

Now, the imu message must contain angle, acceleration and orientation(In other words, it must be a 9-axis IMU.). No covariance is required.

> Last question: How can it be that the modules start running when I specify use_imu: True at moments when i am not launching the IMU? Shouldnt it wait for imu messages to come in before starting?

Beacause even when the IMU is not launched, the modules of lego-loam start running.
This package uses only IMU for distortion correction and does not use imu for preintegration, so I don't think it's a problem to start the module before the imu message comes.

rsasaki0109, 2020-09-05:
Later on I will add a demo using the lego-loam dataset, so please be patient.

jackal_dataset_20170608
https://github.com/RobustFieldAutonomyLab/jackal_dataset_20170608

rsasaki0109, 2020-09-05:
My apologies.
Apparently there was a bug with respect to the IMU, which caused lidar to jump all over the place.

I've just fixed it.

If you set the scan_period to 0.1, it works fine with any of the data in the above dataset.

rsasaki0109, 2022-04-20:
I close this issue because there is no recent activity on this issue.
Please open the issue again if you have any further problems.

closed 2022-04-20 by rsasaki0109

mattdouglas-dt, 2024-12-28:
good afternoon, I am having a similar issue I believe trying to integrate the bno055 IMU into the codeset. even enabling the IMU in the code still doesn't appear to look at the IMU data, and uses the Lidar only for orientation etc, any help would be greatly appreciated in this.

rsasaki0109, 2024-12-29:
The IMU is being used here:
https://github.com/rsasaki0109/lidarslam_ros2/blob/develop/scanmatcher/src/scanmatcher_component.cpp#L241-L243

If you want to use the IMU, this might be a better option:
https://github.com/rsasaki0109/li_slam_ros2

The IMU integration in lidarslam_ros2 is based on LEGO LOAM and is a bit lackluster in some respects, so I suggest using the one above.