rsasaki0109/lidar_slam_ros2 #25: Other 3D LIDAR compatibility with this repository

AvinashRamashray, 2021-03-16:
Hi Sasaki san

I am very new to SLAM with LIDARs.
Please let me know that is this repository is compatible with another 3D-LIDAR or only Velodyne LIDAR is supported.
By the way I have Surestar R-Fans-16 3D LIDAR.
I am able to see in lidarslam.launch.py it is mentioned for velodyne.

If I have to use another LIDAR,is it possible to use it with your repository.

Please confirm.
Looking forward to hear from you soon.

Thanks

rsasaki0109, 2021-03-16:
This package can be used with any rotating 3d lidar other than velodyne. I have also tested this package on robosense and ouster lidar. I don't know about the solid state lidar as I haven't tried it.

AvinashRamashray, 2021-03-16:
Thank you for your information.
Just for information:It seems my LIDAR contains motor and it is rotating.
                                I am able to set its rotating frequency in hertz also. 

 I want to use this repo and  I have some queries:
1. How to integrate any rotating LIDAR to this repository?
   What things we need to do for this?

Can you please give operational steps to be performed in order to achieve SLAM?

Thanks

rsasaki0109, 2021-03-16:
Change the following launch and use it.

Change /velodyne_points in the following to the name of the pointcloud2 type message that the ouster driver publishes.
        remappings=[('/input_cloud','/velodyne_points')],
https://github.com/rsasaki0109/lidarslam_ros2/blob/foxy/lidarslam/launch/lidarslam.launch.py#L21

Then change /velodyne to the frame id of the pointcloud2 type message name that the ouster driver publishes
        arguments=['0','0','0','0','0','0','1','base_link','velodyne']
https://github.com/rsasaki0109/lidarslam_ros2/blob/foxy/lidarslam/launch/lidarslam.launch.py#L28

AvinashRamashray, 2021-03-17:
Thank you for your answer.I will try this.

AvinashRamashray, 2021-04-13:
Hi Sasaki san

I want to use odometry and I am using odom parameter true but getting lookuptransform error as 
[scanmatcher_node-1] [INFO] [1618293549.877689313] [scan_matcher]: Inside try of use_odom_ odom_frame_id= odom,robot_frame_id =base_link
[scanmatcher_node-1] [INFO] [1618293549.877777252] [scan_matcher]: Inside catch of use_odom_ true
[scanmatcher_node-1] [ERROR] [1618293549.877993974] [scan_matcher]: "odom" passed to lookupTransform argument target_frame does not exist. 

My Lidar driver is written in ROS1 and currently I am using ros1_bridge to use topics of ROS1 in ROS2 .
Here to use odometer, TF tree should be like 
                                             map---->odom---->base_link------->lidar_frame.
But actually in rviz2 I am getting TF tree as 
                                            map------->base_link------->lidar_frame.

Please suggest what should be done in this case.

rsasaki0109, 2022-04-18:
Sorry. I'm a little unclear.
It's a tf issue so I think it's better to ask at ros answers.
https://answers.ros.org/questions/

rsasaki0109, 2022-08-18:
I will close this issue at once. Please contact me if you have any further problems.

closed 2022-08-18 by rsasaki0109