rsasaki0109/lidar_slam_ros2 #26: CMake Error: 'tf2_geometry_msgs' was not found before

christophTUM, 2021-04-19:
Hello,
i want to build the package lidarslam_ros2 on Ubuntu 20.04 (Focal Fossa) with ndt_omp_ros2 and g2o installed but i get the following error:

`CMake Error at /opt/ros/foxy/share/ament_cmake_target_dependencies/cmake/ament_target_dependencies.cmake:77 (message):
  ament_target_dependencies() the passed package name 'tf2_geometry_msgs' was
  not found before`

even though the package tf2_geometry_msgs is installed.

rsasaki0109, 2021-04-27:
I see your error, you're probably using a package from the ros2-dashing branch, please use a package from the ros-foxy branch.

ros2 dashing
https://github.com/rsasaki0109/lidarslam_ros2/blob/dashing/scanmatcher/src/scanmatcher_component.cpp#L488
ros2 foxy
https://github.com/rsasaki0109/lidarslam_ros2/blob/foxy/scanmatcher/src/scanmatcher_component.cpp#L488

rsasaki0109, 2021-08-19:
Since there seems to be no response, I will close this isuue. If you have any more problems, please open it again.

closed 2021-08-19 by rsasaki0109