rsasaki0109/lidar_slam_ros2 #28: base_link is jumping.

AvinashRamashray, 2021-05-12:
Hi Sasaki san

Thank you for your help.
I have one query regarding base_link jump.
I am not using odom (flag use_odom = false.) and when I am performing  slam 
base_link is jumping randomly.

For this behavior I am attaching video 
https://www.youtube.com/watch?v=5PK0N8X5Hgg

Bug can be seen after 1:02 minute of video.
Can you please tell what could be the reason for that?
What I am missing here?
============================================================================================

rsasaki0109, 2021-05-12:
Hello.
Probably the ndt scan matching is failing.
Since ndt_num_threads is 2, you might want to increase it, or get the rosbag and run it with rosbag set to x0.5, x0.25, etc.

rsasaki0109, 2021-05-12:
Also, changing the ndt_resolution and voxel grid filter parameters may help.
Is the environment indoors or outdoors?

AvinashRamashray, 2021-05-12:
For now it is indoor........moving my lidar around table

rsasaki0109, 2021-05-12:
Indoors, many of the parameters related to ndt are too large.
For example, the following may improve the behavior. (The following is not the best)
ndt_resolution: 1.0
trans_for_mapupdate: 0.5

Also, this package adds a map every certain distance (trans_for_mapupdate), so if lidar just arounds the table, the map may not be done well.

AvinashRamashray, 2021-05-13:
Thank you for this information.

Used parameters for indoor environment:
ndt_resolution: 1.0
ndt_num_threads: 5
trans_for_mapupdate: 0.2
voxel_leaf_size: 0.1

Result: - I tried above settings and outcome is better than previous result.

So still looking for better results.
I am trying more number of iterations.

AvinashRamashray, 2021-05-27:
Closing this issue.Thanks

closed 2021-05-27 by AvinashRamashray