rsasaki0109/lidar_slam_ros2 #29: TF issue

girvenavery2022, 2021-06-18:
Hello again, 
I am running into some issues with the slam package. when I launch my robot in gazebo/rviz everything checks out, but issues arise when I launch the slam package. The terminal that I launch my robot in and the terminal I launch slam in gives me an error saying: 
          [rviz2-9] Warning: TF_OLD_DATA ignoring data from the past for frame base_footprint at time 289.900000 according to                          authority Authority undetectable
          [rviz2-9] Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
RVIZ also complains saying there is no transformation from Odom to base_link when the slam is running 

The params in the lidarslam are global_frame_id is "map" and robot_frame_id are "base_footprint". I am lost on where to go from here. Am i missing something?

Here are the screenshots of the errors and the tf frames along with the lidarslam package from my local workspace:
[frames.pdf](https://github.com/rsasaki0109/lidarslam_ros2/files/6678740/frames.pdf
[lidarslam_ros2.zip](https://github.com/rsasaki0109/lidarslam_ros2/files/6678742/lidarslam_ros2.zip)
[screenshots.zip](https://github.com/rsasaki0109/lidarslam_ros2/files/6678747/screenshots.zip)

rsasaki0109, 2021-06-18:
Hmm...
Sorry, I'm not familiar with tf and gazebo, so I'm not sure what the cause is.

If you ask a question in ros answer, you may get an answer.
https://answers.ros.org/questions/

rsasaki0109, 2021-06-18:
I used the following rosbag to incorporate odom, and you may want to refer to it.
https://drive.google.com/file/d/1Y6KR9FUQggcyhvGsnkv7zpYQGvc7dQR_/view

rsasaki0109, 2021-06-18:
I used the following rosbag to incorporate odom into this package, and you may want to refer to it.
https://drive.google.com/file/d/1Y6KR9FUQggcyhvGsnkv7zpYQGvc7dQR_/view

girvenavery2022, 2021-06-23:
I was able to resolve that issue but another issue surfaced. I updated some other branches in my work space and now the "num_submaps" are 1 and the "moving distance" is 0 even though i drive my model around. would this be caused by an issue with the libg2o library or a configuration issue with my tf frames or mappings?

rsasaki0109, 2021-06-26:
Is the point cloud being properly input to the lidarslam package?
You might be able to find out what the problem is by setting the debug_flag of scan_matcher to true.
https://github.com/rsasaki0109/lidarslam_ros2/blob/foxy/lidarslam/param/lidarslam.yaml#L28

girvenavery2022, 2021-06-28:
Yeah I had the enabled and the final transformation matrix is reflecting changes from gazebo. I have confirmed it is at least being provided a tf transformation from base_link to laser_link.
I will provide a picture of what the debug flag is showing. 
[image]

rsasaki0109, 2021-06-28:
I think "number of filtered cloud points" is too small, can you show me the yaml file?

girvenavery2022, 2021-06-28:
yeah I can show you the yaml file. I will also include the launch file. in the launch file, I have removed the tf transformation from base_link to lidar link because my robots urdf already provides that transformation. 
[image]
[image]

rsasaki0109, 2021-06-28:
It is recommended to make vg_size_for_input much smaller.

girvenavery2022, 2021-06-28:
I just decreased the vg size and it increase the number of filtered point clouds by a lot but I am still running into the issue.
[image]
[image]

rsasaki0109, 2021-06-28:
0.05 is too small.
A vg_size_input of 0.5, 1.0, etc. is better. (or 0.25?)
Also, is the environment outdoors or indoors?
If indoors, I often use 2.0 for ndt_resolution.
If you can show me a picture of your scan visualization, I may be able to advise you better.

girvenavery2022, 2021-06-28:
I just increased it to .25. The environment the package is used in will be outdoors. yeah i can definitely show you that.
The first picture shows you what rviz2 is displaying from the pointcloud2 topic( /mammoth/filtered_points) and the second picture shows the gazebo world i put together.
[image]
[image]

rsasaki0109, 2021-06-29:
The irradiation range of the lidar is narrow, so if possible, the sensor position should be reviewed to increase the irradiation range more (ideally 360 degrees).
Also, there are few obstacles. This may be causing the scan matching to fail due to the low amount of features.

girvenavery2022, 2021-06-29:
would there be any other causes for the scan matcher to fail? I increased the irradiation range to 360 degrees and added significantly more obstacles and still nothing:(
[image]

girvenavery2022, 2021-06-29:
so i was able to get it working just once. it looked like it updated once while it was running because the num_of_sub maps changed to 13, the distance went to 29m, and the map topic published its point clouds. so its like the scan matcher is getting stuck somewhere or a param is telling it to wait(map_publish_period?), i just don't know what.

rsasaki0109, 2021-06-29:
I checked the parameters, but you might want to increase the ndt_num_thread in scanmatcher.
Also, you may want to adjust vg_size_for_map to 0.05 or 0.1.

rsasaki0109, 2021-06-29:
If you upload the rosbag, I can analyze it.

girvenavery2022, 2021-06-29:
I recorded the topic for the lidar and put it in a rosbag, I also recorded the map topic as well as the modified map topic and put them in a bag. during the recording, the scanmatcher started working for a second so i have 2 bags for the map topic. one for the time it worked and one for the time it didn't work.

rosbag for the lidar topic:
[rosbag2_2021_06_29-17_52_36.zip](https://github.com/rsasaki0109/lidarslam_ros2/files/6736811/rosbag2_2021_06_29-17_52_36.zip)

rosbag for the map that worked:
[rosbag2_2021_06_29-17_47_48.zip](https://github.com/rsasaki0109/lidarslam_ros2/files/6736805/rosbag2_2021_06_29-17_47_48.zip)

rosbag for the map that didn't work:
[rosbag2_2021_06_29-17_52_36.zip](https://github.com/rsasaki0109/lidarslam_ros2/files/6736810/rosbag2_2021_06_29-17_52_36.zip)

rosbag for the modified_map topic:
[rosbag2_2021_06_29-17_48_32.zip](https://github.com/rsasaki0109/lidarslam_ros2/files/6736818/rosbag2_2021_06_29-17_48_32.zip)

rsasaki0109, 2021-07-04:
Sory, I've been busy and don't have a pc for ros2 on hand, so it will take some time to analyze.

girvenavery2022, 2021-07-06:
It’s okay! I’m trying to debug it as well and I’ll let you if I get anything new. Thanks for letting me know though!

girvenavery2022, 2021-09-28:
Hi, I just wanted to let you know I ended up fixing the problem. One of my nodes downstream to the lidarslam node was using system time while the other half of the architecture was using simulation time. so I guess this was upsetting the slam package. Thank you for all of your help