rsasaki0109/lidar_slam_ros2 #39: IMU intergraton fail

cshang412, 2022-04-27:
Everything going smooth with lidar alone mapping , but when set `use_imu: true`,and publish **IMU topic  with 200HZ**, **Lidar 10HZ**, i got error like below:

```
ros2 launch lidarslam lidarslam_tukuba.launch.py 
[INFO] [launch]: All log files can be found below /home/david/.ros/log/2022-04-27-22-21-37-612097-david-pc-1003649
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [scanmatcher_node-1]: process started with pid [1003651]
[INFO] [static_transform_publisher-2]: process started with pid [1003653]
[INFO] [graph_based_slam_node-3]: process started with pid [1003655]
[static_transform_publisher-2] [INFO] [1651069297.746841314] [static_transform_publisher_Bp2zNgpNGCEbeiRh]: Spinning until killed publishing transform from 'base_link' to 'laser_link'
[scanmatcher_node-1] registration_method:NDT
[scanmatcher_node-1] ndt_resolution[m]:2
[scanmatcher_node-1] ndt_num_threads:1
[scanmatcher_node-1] gicp_corr_dist_threshold[m]:5
[scanmatcher_node-1] trans_for_mapupdate[m]:2
[scanmatcher_node-1] vg_size_for_input[m]:1
[scanmatcher_node-1] vg_size_for_map[m]:0.1
[scanmatcher_node-1] use_min_max_filter:true
[scanmatcher_node-1] scan_min_range[m]:1
[scanmatcher_node-1] scan_max_range[m]:100
[scanmatcher_node-1] set_initial_pose:true
[scanmatcher_node-1] use_odom:false
[scanmatcher_node-1] use_imu:true
[scanmatcher_node-1] scan_period[sec]:0.1
[scanmatcher_node-1] debug_flag:false
[scanmatcher_node-1] map_publish_period[sec]:100
[scanmatcher_node-1] num_targeted_cloud:30
[scanmatcher_node-1] ------------------
[scanmatcher_node-1] [INFO] [1651069297.773546138] [scan_matcher]: initialize Publishers and Subscribers
[scanmatcher_node-1] [INFO] [1651069297.781154965] [scan_matcher]: initialization end
[graph_based_slam_node-3] [INFO] [1651069297.788144665] [graph_based_slam]: initialization start
[graph_based_slam_node-3] registration_method:NDT
[graph_based_slam_node-3] voxel_leaf_size[m]:0.2
[graph_based_slam_node-3] ndt_resolution[m]:5
[graph_based_slam_node-3] ndt_num_threads:3
[graph_based_slam_node-3] loop_detection_period[Hz]:5000
[graph_based_slam_node-3] threshold_loop_closure_score:2.5
[graph_based_slam_node-3] distance_loop_closure[m]:100
[graph_based_slam_node-3] range_of_searching_loop_closure[m]:20
[graph_based_slam_node-3] search_submap_num:4
[graph_based_slam_node-3] num_adjacent_pose_cnstraints:5
[graph_based_slam_node-3] use_save_map_in_loop:true
[graph_based_slam_node-3] ------------------
[graph_based_slam_node-3] [INFO] [1651069297.788516988] [graph_based_slam]: initialize Publishers and Subscribers
[graph_based_slam_node-3] [INFO] [1651069297.791048893] [graph_based_slam]: initialization end
[scanmatcher_node-1] [INFO] [1651069301.453226030] [scan_matcher]: create a first map
[scanmatcher_node-1] [pcl::NormalDistributionsTransform::setInputTarget] Invalid or empty point cloud dataset given!
[scanmatcher_node-1] [pcl::VoxelGridCovariance::applyFilter] No input dataset given!
[scanmatcher_node-1] [pcl::registration::NormalDistributionsTransform::compute] No input target dataset was given!
[scanmatcher_node-1] [pcl::registration::NormalDistributionsTransform::compute] No input target dataset was given!
[scanmatcher_node-1] [pcl::registration::NormalDistributionsTransform::compute] No input target dataset was given!
[scanmatcher_node-1] [pcl::registration::NormalDistributionsTransform::compute] No input target dataset was given!
[scanmatcher_node-1] [pcl::registration::NormalDistributionsTransform::compute] No input target dataset was given!
[scanmatcher_node-1] [pcl::registration::NormalDistributionsTransform::compute] No input target dataset was given!
[scanmatcher_node-1] [pcl::registration::NormalDistributionsTransform::compute] No input target dataset was given!
[scanmatcher_node-1] [pcl::registration::NormalDistributionsTransform::compute] No input target dataset was given!
[scanmatcher_node-1] [pcl::registration::NormalDistributionsTransform::compute] No input target dataset was given!
[scanmatcher_node-1] [pcl::registration::NormalDistributionsTransform::compute] No input target dataset was given!
```

Is there any tricks about intergrate IMU to this project?  seems IMU is highly relavant with **time**, should i sync my IMU's time and Lidar's time strictly, Thanks! looking forward for your replying.

rsasaki0109, 2022-05-01:
I can't tell what the problem is just from the terminal output you provided.
Can you provide rosbag?

cshang412, 2022-05-07:
> 

Yes, It's 9-axis IMU , Here is the [ros2 bag ](https://drive.google.com/drive/folders/1jDW1jdpFIF3TQ4poORIGb1IGHzxSULUa)

rsasaki0109, 2022-05-07:
Do you know the tf between imu and lidar? Also, for the imu compound is referring to lego_loam, that could be a reference.
https://github.com/RobustFieldAutonomyLab/LeGO-LOAM

cshang412, 2022-05-08:
> Do you know the tf between imu and lidar? Also, for the imu compound is referring to lego_loam, that could be a reference. https://github.com/RobustFieldAutonomyLab/LeGO-LOAM

Sorry, i forgot about that,
Here is TF(imu is 0.94m lower than lidar)
**extrinsicTrans:  [ 0.0204919,  0.00745582,  -0.94 ]**

rsasaki0109, 2022-05-08:
Fixing the bug about imu in lidarslam_ros2 is going to take a bit of time.

I have confirmed that li_slam_ros2 works with your rosbag, so please use that one.
https://github.com/rsasaki0109/li_slam_ros2

rsasaki0109, 2022-12-08:
I am closing this issue because I did not feel the need to support imu in this package. if you want to use imu, use li_slam_ros2.

closed 2022-12-08 by rsasaki0109