rsasaki0109/lidar_slam_ros2 #40: why g2o setFixed is lastest pose ?

sweetquiet, 2022-05-12:
https://github.com/rsasaki0109/lidarslam_ros2/blob/f11cd369b6b7160e6c413a24585a62b8607ba878/graph_based_slam/src/graph_based_slam_component.cpp#L288

Hi, First ,Thanks for your great work! Why does g2o fix the latest pose instead of the first pose, what is the difference between them? Looking forward to your reply。 thanks。

rsasaki0109, 2022-05-12:
This is for personal reasons, as I didn't want the latest POSE to be discontinuous in graph optimization when applying this package to my robot.

sweetquiet, 2022-05-12:
Thanks for your reply, thanks, I will close this question

closed 2022-05-12 by sweetquiet

rsasaki0109, 2022-09-20:
This specification has been changed due to the following problems.
https://github.com/rsasaki0109/lidarslam_ros2/issues/51