rsasaki0109/lidar_slam_ros2 #42: idea: Scanmathing pose publish with a constraint in roll and pitch.

khalisfadil, 2022-05-12:
Dear Mr. Rsasaki,

I have some question regarding the pose that publish by scanmatching node. I have been thinking about the relation between the limitation and constraint in translation "Z",  orientation "pitch", orientation "roll" with the final pose SLAM output result.

If there is a limit in selected particular movement, can it help to improve the overall SLAM performance.  

to put the constraint in "z" is possible but not with the "pitch" and "roll". I try to convert this line of quaternion into a rotation matrix again. 

line 386 to line 394 (scanmatching component.cpp) void publishMapAndPose

 geometry_msgs::msg::TransformStamped transform_stamped;
  transform_stamped.header.stamp = stamp;
  transform_stamped.header.frame_id = global_frame_id_;
  transform_stamped.child_frame_id = robot_frame_id_;
  transform_stamped.transform.translation.x = position.x();
  transform_stamped.transform.translation.y = position.y();
  transform_stamped.transform.translation.z = position.z();
  transform_stamped.transform.rotation = quat_msg;
  broadcaster_.sendTransform(transform_stamped);

the question is, it is possible? I hope you can give some opinion. Thank you.

rsasaki0109, 2022-05-12:
I think it is possible and not a bad idea, but I do not know if that idea is actually effective.

rsasaki0109, 2022-12-08:
Once this issue is closed. Please comment again if you have any further questions.

closed 2022-12-08 by rsasaki0109