rsasaki0109/lidar_slam_ros2 #43: map->odom transform

felrock, 2022-05-16:
Hello, is it possible to extract map->odom transform with a few changes?

rsasaki0109, 2022-05-16:
This can be done using the lookupTransform function of tf2_ros::Buffer.

felrock, 2022-05-16:
I don't see anywhere in graphbasedslam where you get the correction from adding a pose, usually that is the map transform right?  Also another question if you have the time, both scan-matcher and graph-based-slam have a ndt/gicp registration, why not let scan-matching do scan matching and graph-based-slam only do graph-optimizations?

rsasaki0109, 2022-05-16:
> I don't see anywhere in graphbasedslam where you get the correction from adding a pose, usually that is the map transform right?

Indeed it is. I'll fix it.

> Also another question if you have the time, both scan-matcher and graph-based-slam have a ndt/gicp registration, why not let scan-matching do scan matching and graph-based-slam only do graph-optimizations?

This is confusing my package name, but since scan-matching is lidar odometry with ndt/gicp and graph-based-slam is loop detection with partial map matching and graph optimization with g2o, this configuration is common in other SLAMs.

felrock, 2022-05-19:
Thanks for the info!

closed 2022-05-19 by felrock

oskarmue, 2022-08-17:
Hey sorry I am still a bit confused on how to receive map --> odom.
@felrock did you do this somewhere in your fork, if so I couldnt find it?

felrock, 2022-08-18:
Hi, so map->odom tf is not available, but in short it's the vector that is produced by alignment. So the correction.

So this vector here: https://github.com/rsasaki0109/lidarslam_ros2/blob/991441b9e63eb42e49d44d63a74b61cd4cafbc8e/scanmatcher/src/scanmatcher_component.cpp#L342

is what i believe is the map->odom tf

Please let the author or someone else confirm this first

Destranix, 2023-02-14:
An addition to this topic:
Relying on odom_frame->robot_frame and the publishing global_frame->robot_frame leads to two trees what does cause problems with tf, reporting two unconnected trees on transformation lookup.

The package probably should publish something like global_frame->odom_frame or allow global_frame->somehow_differently_named_node that can be merged by the user with the existing odom_frame ->robot_frame mapping.

Destranix, 2023-02-15:
In addition to my post yesterday:
Publishing map->odom when 'use_odom' is enabled and publishing map->base_link if not, might be a working and sensfull solution.