rsasaki0109/lidar_slam_ros2 #46: Failed to build lidarslam_ros2

SyahirMuzni, 2022-08-30:
System:
- Ubuntu 20.04 Focal Fossa

I was building lidarslam_ros2 with ndt_omp_ros2 installed. But failed to build the scanmatcher & graph_based_slam because there is no g2o. The error is stated as below: 
"CMake Error at CMakeLists.txt:33 (find_package):
  By not providing "Findg2o.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "g2o", but
  CMake did not find one.

  Could not find a package configuration file provided by "g2o" with any of
  the following names:

    g2oConfig.cmake
    g2o-config.cmake

  Add the installation prefix of "g2o" to CMAKE_PREFIX_PATH or set "g2o_DIR"
  to a directory containing one of the above files.  If "g2o" provides a separate development package or SDK, be sure it has been installed."
  
So, I am quite confused because I see in your scanmatcher package stated that I don't have to build the g2o. 
Really appreciate for your help.

rsasaki0109, 2022-08-30:
Install g2o with the following command.
```
sudo apt install ros-foxy-libg2o
```

rsasaki0109, 2022-08-30:
To add, the CI of this package installs the dependent packages with rosdep.
https://github.com/rsasaki0109/lidarslam_ros2/blob/foxy/.github/workflows/main.yml#L40-L41

rsasaki0109, 2022-08-30:
Oh, sorrry. This package is also available for ros2 galactic.
Install g2o with the following command.

```
sudo apt install ros-galactic-libg2o
```

SyahirMuzni, 2022-08-30:
Ok. Thank you. It is done!

SyahirMuzni, 2022-09-01:
Hi.
I tried to run the command 'ros2 launch lidarslam lidarslam.launch.py'

But I encountered this error: 
'Package 'lidarslam' not found: "package 'lidarslam' not found, searching: ['/opt/ros/galactic']'

For your information, I have successfully built the lidarslam package.

Really appreciate your help.

rsasaki0109, 2022-09-01:
In ROS2, the following command must be entered to use a built package.
```
source <Workspace where the package was built>/install/setup.bash
```

SyahirMuzni, 2022-09-07:
Ok thank you for your quick response. I will close this issue

closed 2022-09-07 by SyahirMuzni