rsasaki0109/lidar_slam_ros2 #49: scanmatcher node-1 error after launch

SyahirMuzni, 2022-09-13:
**OS and Version: [Ubuntu 20.04]
ROS Version: [Galactic]**

*The goal:*
To create an outdoor map.

*The issue:*
Error when launched Lidarslam. The error --> [scanmatcher_node-1] [ERROR] [1663050603.996330287] [scan_matcher]: "odom" passed to lookupTransform argument target_frame does not exist.

*Steps to reproduce the behavior:*
- source /opt/ros/galactic/setup.bash
- ros2 launch velodyne velodyne-all-nodes-VLP16-launch.py

On a new tab, rviz2
- Add, by topic pointcloud2
- Fixed frame, velodyne

On a new tab, launching the lidarslam command: 
-source install/setup.bash
-ros2 launch lidarslam lidarslam.launch.py

rsasaki0109, 2022-09-14:
You are not using Odometry, are you?
If use_odom is set to true in the yaml you are using, you need to set it to false.

[lidarslam.yaml](https://github.com/rsasaki0109/lidarslam_ros2/blob/foxy/lidarslam/param/lidarslam.yaml)

SyahirMuzni, 2022-09-14:
Got it! Thank you very much!

closed 2022-09-14 by SyahirMuzni