rsasaki0109/lidar_slam_ros2 #51: Map produced not aligned

SyahirMuzni, 2022-09-19:
Hi!

I have successfully gotten the map data using the lidarslam. But, there is an issue, I noticed that the map is not aligned when I tried to record it on a loop road (you may see the line in the red circle is not aligned). But when I record it on a non-looping road, it works just fine. How can I resolve this issue? 

Thank you.

[image]

rsasaki0109, 2022-09-19:
If you can't provide rosbag, I would appreciate information on the type of lidar you used and the yaml file.

SyahirMuzni, 2022-09-20:
Here is the link of the ros2 bag and yaml file as your requested.
[https://drive.google.com/file/d/1727BwiDbq9OzKh88HWXv_bE_lJSz86sT/view?usp=sharing]
[https://drive.google.com/file/d/1BnLPt1LPNYim5uqHsOd4dShrx96x4_Fo/view?usp=sharing](url)

For your information, I am using Velodyne VLP-16 and the start and end of the loop is at the same point.

rsasaki0109, 2022-09-20:
The url you provided does not seem to be appropriate; clicking on the url will take me to the following

https://github.com/rsasaki0109/lidarslam_ros2/issues/url

lio_bigloop.launch.py makes it easier to close loops over longer distances than this package
But please wait a moment as I will check if this package can also close the loop with your data.

rsasaki0109, 2022-09-20:
I have looked at the data and the distances are long enough to be acceptable for this package.
I think there is a bug in the loop closure, so please wait a little while until it is addressed.

rsasaki0109, 2022-09-20:
The bug has been fixed in the following PR.

[image]
PR
https://github.com/rsasaki0109/lidarslam_ros2/pull/52

SyahirMuzni, 2022-09-20:
Great! I tried to record two loop maps. It worked!

closed 2022-09-20 by SyahirMuzni