rsasaki0109/lidar_slam_ros2 #53: Max distance (km) for mapping

SyahirMuzni, 2022-09-23:
Hi! I have an issue while recording a larger map using *lidarslam.yaml.* The desired route is about **6.2km long.** So, the mapping stopped somewhere while I was recording it [https://drive.google.com/file/d/17_C7tcCxtObeyDzwouvG9QjKfip85lPi/view?usp=sharing](url) 

I believe there is a limit in the yaml file right? If it is so, is it can be solved if I use the *lidarslam_tukuba.yaml*?

rsasaki0109, 2022-09-23:
As long as this package can store the pcd data in memory, it can be mapped with scan matching.
I have only tried loop close with 2-3km of data.

It seems that NDT scan munging is failing. I think lidarslam_tukuba.yaml also fails because the NDT parameters are the same as lidarslam.yam.
Suggested solutions are as follows
Tune NDT parameters (ndt_resolution between 1~5, vg_size_for_input, vg_size_for_map smaller)
Move the robot in a way that it does not rotate as rapidly as possible and does not vibrate too much during the measurement.
Add more odometry and IMU

Provide rosbag so I can pursue the cause in more detail.

SyahirMuzni, 2022-09-29:
I'm sorry I can't provide you the rosbag at the moment. But, those suggestions are make sense to me. Let say I want to add IMU, so I have to use *li_slam* instead of *lidarslam* right?

rsasaki0109, 2022-09-29:
>  I have to use li_slam instead of lidarslam right?

Yes.
> I'm sorry I can't provide you the rosbag at the moment. 

Sad. If you can, give me a video of the rviz of the scan at the time and I might be able to figure out the problem.

SyahirMuzni, 2022-09-29:
I will give you the rosbag soon but not at the moment. I still want to contribute to your work as well as I want the best results for my work.

SyahirMuzni, 2022-10-05:
Hi there,

I have acquired the rosbag for my problem.
[https://drive.google.com/file/d/1Vakv6shABeRXbnqPA83424o1bLOZ4fZ2/view?usp=sharing](url)

**Tools:**
1. Lidar: Velodyne VLP-16
2. GNSS/IMU: XSens MTI-G-710

**Issues:**
1. Lidarslam crashed in the mid-way. 
2. The elevation of the pointcloud is not accurate. Is the data obtains the x, y, z information?

rsasaki0109, 2022-10-06:
In my environment, the scan matching failed occasionally and the loop did not close, but I was able to complete the scan matching.
It will take some time, but I will look into a few things to close the loop.

The reason lidarslam crashed on your PC could possibly be that your PC's CPU memory is small and could not hold the point cloud.
Or maybe the low GPU memory made the rviz/PC run slower, which caused the scan matching process to fail. If you are not visualizing point clouds, GPU memory may not be much of an issue.
For reference, my PC has 32GB of CPU memory and 8GB of GPU memory.

[image]

SyahirMuzni, 2022-10-06:
Here are some more details about the issue. 
[image]

[image]

rsasaki0109, 2022-10-06:
It seems to be the same as my environment. Scan matching is failing in places.

rsasaki0109, 2022-10-17:
I'm sorry. I have been busy lately and it will take some time to get lidarslam_ros2 to close this loop.
If you are in a hurry, please use li_slam_ros2.

KamedaHakobot, 2023-08-23:
>>I want to add IMU, so I have to use li_slam instead of lidarslam right?
>Yes.

Does "use_imu" flag in this package not work now?

rsasaki0109, 2023-08-23:
That feature was inspired by lego-loam, but its performance isn't optimal. If you want to use IMU, it's better to use li_slam.