rsasaki0109/lidar_slam_ros2 #55: Failed   <<< ndt_omp_ros2 [1min 5s, exited with code 2]

OforiDarren, 2022-11-16:
Hi, I'm trying to "colcon build" your humble package that I got from your github. When I try building theapplication I get this error + a lot of code errors as well. Is this code optimized for humble and if so what am I doing wrong since I just cloned the github humble repo.
Version: Ubuntu 22.04, Humble.

[image]
Steps that I took in this process:

1. I switched branches to: https://github.com/rsasaki0109/lidarslam_ros2/tree/feature/humble_ci_and_minor_fix
2. Then I followed the requirement to build steps where you "cd ros2_ws/src" after that I did "git clone --recursive https://github.com/rsasaki0109/lidarslam_ros2"
3.  cd ~/ros2_ws
4. colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
5. rviz2 -d src/lidarslam_ros2/lidarslam/rviz/mapping.rviz 
6. ros2 launch lidarslam lidarslam.launch.py (Doesn't find a package) (Package 'lidarslam' not found: "package 'lidarslam' not found, searching: ['/opt/ros/humble']"
)

rsasaki0109, 2022-11-17:
Is the ~/ros2_ws/src/lidarslam_ros2/Thirdparty/ndt_omp_ros2 branch humble?
https://github.com/rsasaki0109/lidarslam_ros2/tree/humble/Thirdparty
You can check the branch with "git status".

It is also necessary to enter source ~/ros2_ws/install/setup.bash in the terminal to be started before launching the launch file.

OforiDarren, 2022-11-17:
Okay I checked the branche and it gives me this "HEAD detached at af67849" This makes me think that it doesn't have the humble branche since it doesn't match your "[41bdfba]". Should I remove the branche and git clone this one https://github.com/rsasaki0109/ndt_omp_ros2/tree/humble at the same folder location?

rsasaki0109, 2022-11-17:
That's right.

closed 2022-12-08 by rsasaki0109