rsasaki0109/lidar_slam_ros2 #56: Random and silent crashes

Arnwald, 2022-11-18:
I am testing the nodes with point clouds coming from a Robosense RS16, but the nodes keep crashing randomly without any error message. It just stops and I can't figure why.

Is this behavior normal or am I missing something ?

rsasaki0109, 2022-11-18:
I have never encountered that error.
Does your pc have more than 16gb ram?
Also, if you don't mind, could you please provide a rosbag?

Arnwald, 2022-11-18:
Yes, I have 32Go and it does not seem to be the origin of the problem.

I have to add that I am running the nodes with Galactic. I tried various parameters but it does not change anything. scanmatcher and graph_based_slam stop listening and publishing, but they don't die and the cpu usage falls to 0.

Arnwald, 2022-11-22:
Do you need a rosbag of the input data or of the output of the slam node ?

rsasaki0109, 2022-11-24:
I would like it if possible.

Arnwald, 2022-11-24:
Please find the rosbags under this link,
https://www.swisstransfer.com/d/c3ad9dfc-8470-4826-acf5-d91c13b2266c

The compressed folder contains both the rosbag of the input data (provided by the node from Robosense: https://github.com/RoboSense-LiDAR/rslidar_sdk) and the outputs of lidarslam.

Here are the parameters that I used:

scan_matcher:
  ros__parameters:
    global_frame_id: "map"
    robot_frame_id: "base_link"
    registration_method: "NDT"
    ndt_resolution: 2.0
    ndt_num_threads: 0
    gicp_corr_dist_threshold: 5.0
    trans_for_mapupdate: 1.5
    vg_size_for_input: 0.1
    vg_size_for_map: 0.1
    use_min_max_filter: true
    scan_min_range: 5.0
    scan_max_range: 100.0
    scan_period: 0.1
    map_publish_period: 5.0
    num_targeted_cloud: 30
    set_initial_pose: true
    initial_pose_x: 0.0
    initial_pose_y: 0.0
    initial_pose_z: 3.4
    initial_pose_qx: 0.0
    initial_pose_qy: 0.06
    initial_pose_qz: 0.0
    initial_pose_qw: 1.0
    use_imu: false
    use_odom: false
    debug_flag: true

graph_based_slam:
    ros__parameters:
      registration_method: "NDT"
      ndt_resolution: 2.0
      ndt_num_threads: 0
      voxel_leaf_size: 0.2
      loop_detection_period: 1000
      threshold_loop_closure_score: 2.5
      distance_loop_closure: 20.0
      range_of_searching_loop_closure: 20.0
      search_submap_num: 2
      num_adjacent_pose_cnstraints: 5
      use_save_map_in_loop: false
      debug_flag: true

I run the nodes under Ubuntu 20.04 with Galactic.

rsasaki0109, 2022-11-24:
Hmmm. I tried several times, but in my environment, the crash you refer to did not happen. How often did that crash happen to you?
I will make a few more attempts over the weekend.
[image]

rsasaki0109, 2022-11-24:
After looking at rosbag2_2022_11_22-16_12_36, I probably found the cause.
I think the scan matching failed due to temporarily heavy processing load.

If you take the following actions, scan matching will be less likely to fail.
・Reduce the playback rate of rosbag( `-r X` option)
・Do not do any other work during SLAM on your PC.
・Adjust ndt_num_threads to the number of threads on your PC.
・Turn off rviz.

Arnwald, 2022-11-25:
I already tried reducing ndt_num_threads or turning rviz off, but unfortunately it didn't help. I am monitoring the load on the PC and it does not exceed 30% of the total computing power (this is an 8 core i7). Reducing the rate of rosbag doesn't help either.

rsasaki0109, 2022-11-27:
Hmmm.
I don't know the cause of the problem.
By the way, how often did the crash happen to you?

Arnwald, 2022-11-27:
It happens every time I launch it, but after a random time. I don’t understand what is happening either.
I also tried the package LIO-SAM and the exact same problem occurred.

rsasaki0109, 2022-11-27:
Hmmm. The cause seems to be your ubuntu environment or PC, but I am not very knowledgeable in that area.
You might want to change your PC or reinstall ubuntu.

Arnwald, 2022-11-27:
I also tried on another PC and I had the same problem. But the PC was the exact same model as mine, so I am planning to try it on a third PC.
I’ll keep you posted.

Arnwald, 2022-11-30:
So, I finally found the problem. I was launching the slam nodes and the rosbag separately. I modified the launch.py file to launch the slam nodes, the rosbag and rviz from it, and now everything runs as it should.
Maybe there is something that I don't understand with ROS2, or there is a bug in the messages handling with this kind of usage.

rsasaki0109, 2022-12-08:
I think the issue has been resolved, so I will close this issue. I don't have a deep understanding of ros2, so if you have questions, I suggest you ask ros answers.

closed 2022-12-08 by rsasaki0109

Arnwald, 2022-12-08:
No problem. Thank you again.