rsasaki0109/lidar_slam_ros2 #58: colcon installation failure deu to opengl dependency

ShaharSarShalom, 2022-11-22:
Hi 

I'm trying to install the graph_based_slam_node 

Colcon keeps on failing due to opengl dependency. I tried installing the opengl libaray, nothing helped. 

Here is how I run the colcon command - 

`colcon build  --symlink-install --parallel-workers 12 --ament-cmake-args -DPYTHON_EXECUTABLE=$MY_PYTHON_ENV -DCMAKE_BUILD_TYPE=Release`

`CMake Error at CMakeLists.txt:69 (add_executable):
  Target graph_based_slam_node links to target OpenGL::GL but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at CMakeLists.txt:69 (add_executable):
  Target graph_based_slam_node links to target OpenGL::GLU but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?`

rsasaki0109, 2022-11-24:
I'm not sure what the cause is, since graph_based_slam_node has no OpenGL dependencies.
https://github.com/rsasaki0109/lidarslam_ros2/blob/foxy/graph_based_slam/CMakeLists.txt

Is it foxy in your ros environment? Or humble?
The branch of lidarslam_ros2 you are using matches your ros environment?
Also, is your OS ubuntu?

ShaharSarShalom, 2022-11-24:
I experience environmental issues. 
I work on Ubuntu 20.04, foxy. python 3.8

I cloned the default repository, I believe it would be foxy 
I believe there is something in my environment, not your code

closed 2022-11-24 by ShaharSarShalom