rsasaki0109/lidar_slam_ros2 #60: About the timeline of map creation

g2181053, 2022-12-21:
Perhaps a rudimentary question, but...
Is it possible to create a map from a previously recorded BAG file?

Or would it create a map from records from LIDAR in the ongoing process?

Looking forward to your reply！

rsasaki0109, 2022-12-21:
It is possible to create a map from a previously recorded BAG file.

g2181053, 2022-12-21:
Thank you for your reply.

I immediately tried the following method. (already built with 'ros2_ws' file)

1, I set use_odom in ros2_ws/src/lidarslam_ros2/lidarslam/param/lidarslam.yaml to false and saved it.

2, in the terminal
$ source /opt/ros/galactic/setup.bash
$ rviz2

**3, Open another terminal and run
$ cd ~/ros2_ws
$ . install/local_setup.bash
$ ros2 launch lidarslam lidarslam.launch.py**

4, Open another terminal and run
$ source /opt/ros/galactic/setup.bash
$ ros2 bag play rosbag2_2022_12_06-22_59_36

However, the changes in ros2_ws/src/lidarslam_ros2/lidarslam/param/lidarslam.yaml do not seem to have been reflected, with the following results

$ ros2 launch lidarslam lidarslam.launch.py

[INFO] [launch]: All log files can be found below /home/seiken/.ros/log/2022-12-21-16-16-37-738761-seiken-GL65-Leopard-10SFR-24084
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [scanmatcher_node-1]: process started with pid [24086]
[INFO] [static_transform_publisher-2]: process started with pid [24088]
[INFO] [graph_based_slam_node-3]: process started with pid [24090]
[static_transform_publisher-2] [INFO] [1671606997.924050501] [static_transform_publisher_1TgacbW9MgGvUekq]: Spinning until killed publishing transform from 'base_link' to 'velodyne'
[scanmatcher_node-1] registration_method:NDT
[scanmatcher_node-1] ndt_resolution[m]:2
[scanmatcher_node-1] ndt_num_threads:2
[scanmatcher_node-1] gicp_corr_dist_threshold[m]:5
[scanmatcher_node-1] trans_for_mapupdate[m]:1.5
[scanmatcher_node-1] vg_size_for_input[m]:0.5
[scanmatcher_node-1] vg_size_for_map[m]:0.1
[scanmatcher_node-1] use_min_max_filter:true
[scanmatcher_node-1] scan_min_range[m]:1
[scanmatcher_node-1] scan_max_range[m]:200
[scanmatcher_node-1] set_initial_pose:true
**[scanmatcher_node-1] use_odom:true**
[scanmatcher_node-1] use_imu:false
[scanmatcher_node-1] scan_period[sec]:0.2
[scanmatcher_node-1] debug_flag:false
[scanmatcher_node-1] map_publish_period[sec]:15
[scanmatcher_node-1] num_targeted_cloud:20
[scanmatcher_node-1] ------------------
[scanmatcher_node-1] [INFO] [1671606997.934610910] [scan_matcher]: initialize Publishers and Subscribers
[scanmatcher_node-1] [INFO] [1671606997.937177475] [scan_matcher]: initialization end
[graph_based_slam_node-3] [INFO] [1671606997.953965609] [graph_based_slam]: initialization start
[graph_based_slam_node-3] registration_method:NDT
[graph_based_slam_node-3] voxel_leaf_size[m]:0.1
[graph_based_slam_node-3] ndt_resolution[m]:1
[graph_based_slam_node-3] ndt_num_threads:2
[graph_based_slam_node-3] loop_detection_period[Hz]:3000
[graph_based_slam_node-3] threshold_loop_closure_score:0.7
[graph_based_slam_node-3] distance_loop_closure[m]:100
[graph_based_slam_node-3] range_of_searching_loop_closure[m]:20
[graph_based_slam_node-3] search_submap_num:2
[graph_based_slam_node-3] num_adjacent_pose_cnstraints:5
[graph_based_slam_node-3] use_save_map_in_loop:true
[graph_based_slam_node-3] debug_flag:true
[graph_based_slam_node-3] ------------------
[graph_based_slam_node-3] [INFO] [1671606997.954254737] [graph_based_slam]: initialize Publishers and Subscribers
[graph_based_slam_node-3] [INFO] [1671606997.955671796] [graph_based_slam]: initialization end
[scanmatcher_node-1] [INFO] [1671607005.256040772] [scan_matcher]: create a first map
[scanmatcher_node-1] [ERROR] [1671607005.258227925] [scan_matcher]: "odom" passed to lookupTransform argument target_frame does not exist. 
[scanmatcher_node-1] [ERROR] [1671607005.493541445] [scan_matcher]: "odom" passed to lookupTransform argument target_frame does not exist. 
[scanmatcher_node-1] [ERROR] [1671607005.648704366] [scan_matcher]: "odom" passed to lookupTransform argument target_frame does not exist. 
[scanmatcher_node-1] [ERROR] [1671607005.811000569] [scan_matcher]: "odom" passed to lookupTransform argument target_frame does not exist. 
[scanmatcher_node-1] [ERROR] [1671607005.864350985] [scan_matcher]: "odom" passed to lookupTransform argument target_frame does not exist.

Omitted below

Is there anything I can do to help?

rsasaki0109, 2022-12-21:
@g2181053 
In ros2, parameter changes are not reflected until colcon build.

g2181053, 2022-12-22:
返信ありがとうございます。
エラーコードは出なくなりました！！

質問なのですが、mapを保存するにはどうすればよいのでしょうか？

$ ros2 launch lidarslam lidarslam_tukuba.launch.py
を実行したあとに、それぞれ別のターミナルでオリジナルのbagファイルをplay、rviz2を実行しました。

確かに、rviz上でpathやmapが追加されていることを確認しましたが、
それらを選択してもなにも映りません。

なにかわかりますでしょうか？

rsasaki0109, 2022-12-22:
@g2181053 
README.mdにも書いてありますが、下記のコマンドを実行して下さい。
```
ros2 service call /map_save std_srvs/Empty
```

g2181053, 2023-01-13:
申し訳ないです。いろいろ調べた結果、原因がわかったかもしれません。

記録したbagファイルを再生し、rviz2で見てみると
frame id がmapの時に、fixed frame does not exist と表示されていました!!

勉強した感じでは、mapの原点をどうにかして決定する必要があるとおもうのですが、
方法がなかなかわかりません。

なにか解決策がわかりますでしょうか？

お忙しいなか恐縮ですが、返信お待ちしております。

rsasaki0109, 2023-01-13:
そもそもSLAMがちゃんと回っているのは確認しましたか？
ターミナル出力はどうなっていますか？

rsasaki0109, 2023-01-13:
まだ地図が取得できていないんですかね？
上記のコマンドを起動したディレクトリかパッケージをビルドしたワークスペースに地図ができていませんか？

rsasaki0109, 2023-01-14:
ターミナル出力はどうなっていますか？
よくある問題として、「入力点群のトピック名が間違っている」、「オドメトリーを使っていないのにyamlのuse_odomがtrueになっている」が挙げられます。
後者に関してはデフォルトはuse_odomがfalseになるように修正しときます。

g2181053, 2023-01-14:
申し訳ないです。slamが回っているか確認する方法がいまいちわかりません。

自分のやったこととして、velodyne lidarで記録したbagファイルをplayし、
別ターミナルで $ ros2 launch lidarslam lidarslam_tukuba.launch.pyを実行 (odomはfalseになっている)
さらに別ターミナルでrviz2を実行したところ、By topicにmodified_pathトピックやpoints_rawトピックが追加されていることは確認できましたが、それらを選択してもなにも映らないです・・

なお、$  ros2 launch lidarslam lidarslam_tukuba.launch.pyを実行した結果は以下のとおりでした。

~/ros2_ws$  ros2 launch lidarslam lidarslam_tukuba.launch.py
[INFO] [launch]: All log files can be found below /home/seiken/.ros/log/2023-01-14-19-51-29-272973-seiken-GL65-Leopard-10SFR-39103
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [scanmatcher_node-1]: process started with pid [39105]
[INFO] [graph_based_slam_node-2]: process started with pid [39107]
[INFO] [static_transform_publisher-3]: process started with pid [39109]
[static_transform_publisher-3] [INFO] [1673693489.476176879] [static_transform_publisher_lr9tAwQIQzbDP2VK]: Spinning until killed publishing transform from 'base_link' to 'velodyne'
[scanmatcher_node-1] [WARN] [1673693489.476556587] [transform_listener_impl_55a47b8a9ad8]: New publisher discovered on topic '/tf_static', offering incompatible QoS. No messages will be sent to it. Last incompatible policy: DURABILITY_QOS_POLICY
[scanmatcher_node-1] registration_method:NDT
[scanmatcher_node-1] ndt_resolution[m]:2
[scanmatcher_node-1] ndt_num_threads:1
[scanmatcher_node-1] gicp_corr_dist_threshold[m]:5
[scanmatcher_node-1] trans_for_mapupdate[m]:2
[scanmatcher_node-1] vg_size_for_input[m]:1
[scanmatcher_node-1] vg_size_for_map[m]:0.1
[scanmatcher_node-1] use_min_max_filter:true
[scanmatcher_node-1] scan_min_range[m]:1
[scanmatcher_node-1] scan_max_range[m]:100
[scanmatcher_node-1] set_initial_pose:true
[scanmatcher_node-1] use_odom:false
[scanmatcher_node-1] use_imu:false
[scanmatcher_node-1] scan_period[sec]:0.2
[scanmatcher_node-1] debug_flag:false
[scanmatcher_node-1] map_publish_period[sec]:100
[scanmatcher_node-1] num_targeted_cloud:30
[scanmatcher_node-1] ------------------
[scanmatcher_node-1] [INFO] [1673693489.477702621] [scan_matcher]: initialize Publishers and Subscribers
[scanmatcher_node-1] [INFO] [1673693489.479471558] [scan_matcher]: initialization end
[graph_based_slam_node-2] [INFO] [1673693489.495860869] [graph_based_slam]: initialization start
[graph_based_slam_node-2] registration_method:NDT
[graph_based_slam_node-2] voxel_leaf_size[m]:0.2
[graph_based_slam_node-2] ndt_resolution[m]:5
[graph_based_slam_node-2] ndt_num_threads:1
[graph_based_slam_node-2] loop_detection_period[Hz]:5000
[graph_based_slam_node-2] threshold_loop_closure_score:2.5
[graph_based_slam_node-2] distance_loop_closure[m]:100
[graph_based_slam_node-2] range_of_searching_loop_closure[m]:20
[graph_based_slam_node-2] search_submap_num:4
[graph_based_slam_node-2] num_adjacent_pose_cnstraints:5
[graph_based_slam_node-2] use_save_map_in_loop:true
[graph_based_slam_node-2] debug_flag:true
[graph_based_slam_node-2] ------------------
[graph_based_slam_node-2] [INFO] [1673693489.496283565] [graph_based_slam]: initialize Publishers and Subscribers
[graph_based_slam_node-2] [INFO] [1673693489.498676575] [graph_based_slam]: initialization end
[graph_based_slam_node-2] [WARN] [1673693489.502917854] [transform_listener_impl_55de53865990]: New publisher discovered on topic '/tf_static', offering incompatible QoS. No messages will be sent to it. Last incompatible policy: DURABILITY_QOS_POLICY

自分の説明がヘタで申し訳ないです。

rsasaki0109, 2023-01-15:
下記をちゃんとrosbag内のvelodyeneのscan名に指定しましたか？また、velodyneのトピックはちゃんとPointCloud2型ですか？
https://github.com/rsasaki0109/lidarslam_ros2/blob/humble/lidarslam/launch/lidarslam_tukuba.launch.py#L21

g2181053, 2023-01-17:
返信ありがとうございます。

結果から言いますと、mapの作成に成功いたしました！！本当にありがとうございます。

原因は、sasaki様の言ったてましたとおり
https://github.com/rsasaki0109/lidarslam_ros2/blob/humble/lidarslam/launch/lidarslam_tukuba.launch.py#L21
の記述が間違っておりました。

☓　remappings=[('/input_cloud','/points_raw')],
○　remappings=[('/input_cloud','/velodyne_points')],

lidarslam.launch.pyを参考にlidarslam_tukuba.launch.pyを上記の通りに変更しましたところ、
正常に動作いたしました。

本当にありがとうございました！！

g2181053, 2023-01-17:
返信ありがとうございます。

結果から言いますと、mapの作成に成功いたしました！！本当にありがとうございます。

原因は、sasaki様の言ったてましたとおり
https://github.com/rsasaki0109/lidarslam_ros2/blob/humble/lidarslam/launch/lidarslam_tukuba.launch.py#L21
の記述が間違っておりました。

☓　remappings=[('/input_cloud','/points_raw')],
○　remappings=[('/input_cloud','/velodyne_points')],

lidarslam.launch.pyを参考にlidarslam_tukuba.launch.pyを上記の通りに変更しましたところ、
正常に動作いたしました。

本当にありがとうございました！！
[image]

rsasaki0109, 2023-01-17:
問題は解決したようなのでこのissueは閉じます。

closed 2023-01-17 by rsasaki0109