rsasaki0109/lidar_slam_ros2 #61: Build issue

cjffly, 2022-12-30:
OS: Pop!_OS 22.04 (ubuntu 22.04)
ROS: humble

Hi, I got an issue while building the package:

```
Could not find a package configuration file provided by "g2o" with any of
  the following names:

    g2oConfig.cmake
    g2o-config.cmake

```

After I installed the g2o lib(https://github.com/RainerKuemmerle/g2o), and try to build again, it then reports:

```
Starting >>> scanmatcher
--- stderr: graph_based_slam                                                                    
** WARNING ** io features related to pcap will be disabled
CMake Error at /usr/local/lib/cmake/g2o/modules/FindSuiteSparse.cmake:182 (message):
  Failed to find SuiteSparse - Did not find AMD header (required SuiteSparse
  component).
Call Stack (most recent call first):
  /usr/local/lib/cmake/g2o/modules/FindSuiteSparse.cmake:232 (suitesparse_report_not_found)
  /usr/local/lib/cmake/g2o/modules/FindSuiteSparse.cmake:323 (suitesparse_find_component)
  /usr/share/cmake-3.22/Modules/CMakeFindDependencyMacro.cmake:47 (find_package)
  /usr/local/lib/cmake/g2o/g2oConfig.cmake:7 (find_dependency)
  CMakeLists.txt:38 (find_package)
```

What could be the reason for this error?

cjffly, 2022-12-30:
[update]
using rosdep solved the problem.

```
sudo rosdep init
rosdep update
```
and then 
```
rosdep install --from-paths src -y --ignore-src
```

rsasaki0109, 2022-12-31:
Thanks for the report, I've added the rosdep command to the readme.md.
The issue seems to be resolved, so I will cloze this issue.

closed 2022-12-31 by rsasaki0109