rsasaki0109/lidar_slam_ros2 #62: I had an error when I execute colcon build

HasanEnesBagci, 2023-01-24:
Hello There!
I have an error when I execute:
 `colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release`
I get output:
`Starting >>> tf2     
Starting >>> tf2_msgs
Starting >>> lidarslam_msgs
Starting >>> ndt_omp_ros2
Finished <<< tf2 [0.53s]                                                                                                                                
Starting >>> tf2_py
Starting >>> tf2_eigen_kdl
Finished <<< ndt_omp_ros2 [0.63s]                                                                                                                                                         
Finished <<< tf2_py [0.21s]                                                                                                                           
Finished <<< tf2_eigen_kdl [0.22s]
Finished <<< lidarslam_msgs [0.77s]
Finished <<< tf2_msgs [0.85s]                   
Starting >>> tf2_ros
Finished <<< tf2_ros [0.26s]                      
Starting >>> tf2_geometry_msgs
Starting >>> tf2_eigen
Starting >>> tf2_sensor_msgs
Starting >>> tf2_kdl
Starting >>> tf2_bullet
Starting >>> tf2_tools
Starting >>> examples_tf2_py
Finished <<< tf2_sensor_msgs [0.36s]                                                                                                                                                                     
Finished <<< tf2_tools [0.36s]                                                                                                                                                                           
Finished <<< tf2_bullet [0.38s]
Finished <<< tf2_eigen [0.40s]
Finished <<< tf2_kdl [0.40s]
Finished <<< tf2_geometry_msgs [0.42s]
Starting >>> graph_based_slam
Starting >>> scanmatcher
Starting >>> geometry2
Starting >>> test_tf2
Finished <<< geometry2 [0.12s]                                                                                                                                          
Finished <<< test_tf2 [0.18s]                                                                                                                            
--- stderr: scanmatcher
In file included from /home/hasanenesbagci/ros2_ws/src/lidarslam_ros2/scanmatcher/src/scanmatcher_component.cpp:1:
/home/hasanenesbagci/ros2_ws/src/lidarslam_ros2/scanmatcher/include/scanmatcher/scanmatcher_component.h:47:10: fatal error: tf2_sensor_msgs/tf2_sensor_msgs.hpp: No such file or directory
   47 | #include <tf2_sensor_msgs/tf2_sensor_msgs.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/scanmatcher_component.dir/build.make:63: CMakeFiles/scanmatcher_component.dir/src/scanmatcher_component.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:80: CMakeFiles/scanmatcher_component.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< scanmatcher [0.22s, exited with code 2]
Aborted  <<< graph_based_slam [0.23s]
Aborted  <<< examples_tf2_py [0.77s]                       

Summary: 15 packages finished [2.10s]
  1 package failed: scanmatcher
  2 packages aborted: examples_tf2_py graph_based_slam
  2 packages had stderr output: graph_based_slam scanmatcher
  1 package not processed
`
Can you help me about this problem ?

rsasaki0109, 2023-01-25:
Have you done "rosdep install"?

https://github.com/rsasaki0109/lidarslam_ros2#requirement-to-build

Once you delete build/log/install and then build the package, you may be able to build it.

HasanEnesBagci, 2023-01-26:
> Have you done "rosdep install"?
> 
> https://github.com/rsasaki0109/lidarslam_ros2#requirement-to-build
> 
> Once you delete build/log/install and then build the package, you may be able to build it.

I tried and get an error like below. Also I have installed pcl 1.12 but still I get error.
```usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:719:37: error: ‘class Eigen::Block<Eigen::Matrix<int, -1, -1>, 3, 13, false>’ has no member named ‘cols’; did you mean ‘cos’?
  719 |   if(((dst.rows()!=dstRows) || (dst.cols()!=dstCols)))
      |                                 ~~~~^~~~
      |                                 cos
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h: In instantiation of ‘void Eigen::internal::resize_if_allowed(DstXprType&, const SrcXprType&, const Eigen::internal::assign_op<T1, T2>&) [with DstXprType = Eigen::Block<Eigen::Matrix<int, -1, -1>, 3, 13, false>; SrcXprType = Eigen::CwiseUnaryOp<Eigen::internal::scalar_opposite_op<int>, const Eigen::Matrix<int, -1, -1> >; T1 = int; T2 = int]’:
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:734:20:   required from ‘void Eigen::internal::call_dense_assignment_loop(DstXprType&, const SrcXprType&, const Functor&) [with DstXprType = Eigen::Block<Eigen::Matrix<int, -1, -1>, 3, 13, false>; SrcXprType = Eigen::CwiseUnaryOp<Eigen::internal::scalar_opposite_op<int>, const Eigen::Matrix<int, -1, -1> >; Functor = Eigen::internal::assign_op<int, int>]’
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:879:31:   required from ‘static void Eigen::internal::Assignment<DstXprType, SrcXprType, Functor, Eigen::internal::Dense2Dense, Weak>::run(DstXprType&, const SrcXprType&, const Functor&) [with DstXprType = Eigen::Block<Eigen::Matrix<int, -1, -1>, 3, 13, false>; SrcXprType = Eigen::CwiseUnaryOp<Eigen::internal::scalar_opposite_op<int>, const Eigen::Matrix<int, -1, -1> >; Functor = Eigen::internal::assign_op<int, int>; Weak = void]’
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:836:49:   required from ‘void Eigen::internal::call_assignment_no_alias(Dst&, const Src&, const Func&) [with Dst = Eigen::Block<Eigen::Matrix<int, -1, -1>, 3, 13, false>; Src = Eigen::CwiseUnaryOp<Eigen::internal::scalar_opposite_op<int>, const Eigen::Matrix<int, -1, -1> >; Func = Eigen::internal::assign_op<int, int>]’
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:804:27:   required from ‘void Eigen::internal::call_assignment(Dst&, const Src&, const Func&, typename Eigen::internal::enable_if<(! Eigen::internal::evaluator_assume_aliasing<Src>::value), void*>::type) [with Dst = Eigen::Block<Eigen::Matrix<int, -1, -1>, 3, 13, false>; Src = Eigen::CwiseUnaryOp<Eigen::internal::scalar_opposite_op<int>, const Eigen::Matrix<int, -1, -1> >; Func = Eigen::internal::assign_op<int, int>; typename Eigen::internal::enable_if<(! Eigen::internal::evaluator_assume_aliasing<Src>::value), void*>::type = void*]’
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:782:18:   required from ‘void Eigen::internal::call_assignment(Dst&, const Src&) [with Dst = Eigen::Block<Eigen::Matrix<int, -1, -1>, 3, 13, false>; Src = Eigen::CwiseUnaryOp<Eigen::internal::scalar_opposite_op<int>, const Eigen::Matrix<int, -1, -1> >]’
/usr/include/eigen3/Eigen/src/Core/Assign.h:66:28:   required from ‘Derived& Eigen::MatrixBase<Derived>::operator=(const Eigen::DenseBase<OtherDerived>&) [with OtherDerived = Eigen::CwiseUnaryOp<Eigen::internal::scalar_opposite_op<int>, const Eigen::Matrix<int, -1, -1> >; Derived = Eigen::Block<Eigen::Matrix<int, -1, -1>, 3, 13, false>]’
/usr/local/include/pcl-1.12/pcl/filters/voxel_grid.h:125:54:   required from here
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:719:12: error: ‘class Eigen::Block<Eigen::Matrix<int, -1, -1>, 3, 13, false>’ has no member named ‘rows’; did you mean ‘row’?
  719 |   if(((dst.rows()!=dstRows) || (dst.cols()!=dstCols)))
      |        ~~~~^~~~
      |        row
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:719:37: error: ‘class Eigen::Block<Eigen::Matrix<int, -1, -1>, 3, 13, false>’ has no member named ‘cols’; did you mean ‘cos’?
  719 |   if(((dst.rows()!=dstRows) || (dst.cols()!=dstCols)))
      |                                 ~~~~^~~~
      |                                 cos
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h: In instantiation of ‘void Eigen::internal::resize_if_allowed(DstXprType&, const SrcXprType&, const Eigen::internal::assign_op<T1, T2>&) [with DstXprType = Eigen::Matrix<float, 4, 1>; SrcXprType = Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<float>, Eigen::Matrix<float, 4, 1> >; T1 = float; T2 = float]’:
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:734:20:   required from ‘void Eigen::internal::call_dense_assignment_loop(DstXprType&, const SrcXprType&, const Functor&) [with DstXprType = Eigen::Matrix<float, 4, 1>; SrcXprType = Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<float>, Eigen::Matrix<float, 4, 1> >; Functor = Eigen::internal::assign_op<float, float>]’
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:879:31:   required from ‘static void Eigen::internal::Assignment<DstXprType, SrcXprType, Functor, Eigen::internal::Dense2Dense, Weak>::run(DstXprType&, const SrcXprType&, const Functor&) [with DstXprType = Eigen::Matrix<float, 4, 1>; SrcXprType = Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<float>, Eigen::Matrix<float, 4, 1> >; Functor = Eigen::internal::assign_op<float, float>; Weak = void]’
[... cut ...]
/usr/local/include/pcl-1.12/pcl/point_representation.h:312:40:   required from ‘void pcl::DefaultFeatureRepresentation<PointDefault>::copyToFloatArray(const PointDefault&, float*) const [with PointDefault = pcl::PPFSignature]’
/usr/local/include/pcl-1.12/pcl/point_representation.h:310:7:   required from here
/usr/local/include/pcl-1.12/pcl/point_representation.h:263:88: error: ‘value’ is not a member of ‘pcl::traits::offset<pcl::PPFSignature, pcl::fields::alpha_m>’
make[2]: *** [CMakeFiles/ndt_omp.dir/build.make:63: CMakeFiles/ndt_omp.dir/src/pclomp/voxel_grid_covariance_omp.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:107: CMakeFiles/ndt_omp.dir/all] Error 2
```

rsasaki0109, 2023-01-26:
Why did you think you could build by installing the latest pcl?

Your environment is broken because the build on CI is passing.
I recommend you to build in a clean environment referring to CI.
https://github.com/rsasaki0109/lidarslam_ros2/blob/humble/.github/workflows/main.yml
Note that the branch to clone depends on the ros distribution.

HasanEnesBagci, 2023-01-26:
> 

Launch works but when I entered `ros2 bag play  hdl_501_filtered.bag` without install rosbag_v2 plugin it gives an error: 
```[ERROR] [1674844034.014774548] [rosbag2_storage]: Could not open 'hdl_501_filtered.bag' with 'sqlite3'. Error: Failed to setup storage. Error: Error when processing SQL statement. SQLite error (26): file is not a database
[ERROR] [1674844034.014798312] [rosbag2_storage]: Could not load/open plugin with storage id 'sqlite3'.
[ERROR] [1674844034.014812310] [rosbag2_transport]: Failed to play: No storage could be initialized. Abort
```

rsasaki0109, 2023-01-26:
Install the rosbag_v2 plugin.

rsasaki0109, 2023-01-26:
There is also a way to convert ros1 rosbag to ros2 rosbag with the following package.

https://pypi.org/project/rosbags/

HasanEnesBagci, 2023-01-28:
I tried and I get these
I think I failed to install 
[image]

rsasaki0109, 2023-01-28:
Are you sure you installed rosbag2-bag-v2-plugins correctly?
What is the output when the following command is entered?

sudo apt install -y ros-foxy-rosbag2-bag-v2-plugins

rsasaki0109, 2023-01-30:
It might be better to use rosbags.

HasanEnesBagci, 2023-01-31:
sorry for the latency I had to reinstall ubuntu so all my files is gone , if I try to open the bag again I'll try to contact with you, thx for the responses.

closed 2023-01-31 by HasanEnesBagci