rsasaki0109/lidar_slam_ros2 #63: Ver

remauxl, 2023-02-11:
Hello, after I successfully completed the requirements to build steps, I encountered the fault that is below in the last step which is for using to build. What should I do?

`colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
`

```

user@ubuntu:~/ros2_ws$ colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release

Starting >>> lidarslam_msgs
Starting >>> ndt_omp_ros2
Finished <<< lidarslam_msgs [0.72s]                                                             
--- stderr: ndt_omp_ros2                         
CMake Error at CMakeLists.txt:23 (find_package):
  Could not find a configuration file for package "PCL" that is compatible
  with requested version "1.12".

  The following configuration files were considered but not accepted:

    /usr/share/pcl-1.8/PCLConfig.cmake, version: 1.8.0

make: *** [Makefile:370: cmake_check_build_system] Error 1
---
Failed   <<< ndt_omp_ros2 [0.96s, exited with code 2]

Summary: 1 package finished [1.23s]
  1 package failed: ndt_omp_ros2
  1 package had stderr output: ndt_omp_ros2
  3 packages not processed

```

rsasaki0109, 2023-02-12:
Depending on the version of ros2 (dashing/foxy/melodic), you will need to switch branches for this package.

remauxl, 2023-02-12:
It's 22.04 LTS Ubuntu and I installed the latest version of foxy from ros.org/en/foxy 2 days ago. Should I try with other distributions ?
[image]

rsasaki0109, 2023-02-12:
For foxy, it is the following branch.
https://github.com/rsasaki0109/lidarslam_ros2/tree/foxy