rsasaki0109/lidar_slam_ros2 #67: Data Format: Compatibility/Conversion to NAV2 usage

marcusvinicius178, 2023-04-17:
Hi does this mapping framework provide output data to be used for Navigation Stack2 consumption?
In Nav2, it is used nav_msgs/**OccupancyGrid** map, if I am not wrong (Arrays from 0 to 100 to assign occupancy).

This framework provides: /map_array(lidarslam_msgs/MapArray) and /modified_map (sensor_msgs/PointCloud2)

Is there some available pkg to integrate this map to NAV2 Slam-toolbox usage?

Another question is if this mapping manner is more reliable and efficient than using occupancyGrid map. My application is also outdoor off-road. For this reason, it seems more reasonable to use this lidarslam_ros2 toolbox, it seems more faithful to the mapped environment. But I am not sure how to use it alongside NAV2.

I will be thankful for any hint (application pkg example) in this regard.

rsasaki0109, 2023-04-17:
I do not provide output data in the OccupancyGrid format for use with Nav2 (Navigation Stack2) in this mapping framework. The framework provides /map_array (lidarslam_msgs/MapArray) and /modified_map (sensor_msgs/PointCloud2).

I am not familiar with methods for combining Nav2 with 3D point cloud maps, and I cannot confirm whether there are existing packages for this purpose. However, considering that your application is for outdoor off-road environments, using this lidarslam_ros2 toolbox may enable more realistic and faithful mapping to the environment, but you might also need to improve the 2D Nav2 side.

I cannot provide specific hints or application package examples for using this framework alongside Nav2. However, you may be able to obtain advice from other developers in community forums or discussion boards.

marcusvinicius178, 2023-04-18:
Got it! This map is used within Autoware or also other stacks which offers slam+navigation? 

I will check with other developers on discord. Thanks for the feedback.

closed 2023-04-18 by marcusvinicius178

rsasaki0109, 2023-04-18:
Autoware requires a 3D point cloud map for localization, and maps created with this package can also be used.