rsasaki0109/lidar_slam_ros2 #69: scanmatcher_node-1 process has died

Tester2009, 2023-05-30:
Hi all.

Currently I'm using ROS2 humble, and using LS Lidar C32. I have recorded a pointcloud in a ros bag and try to play it back in our office.

First I run lidarslam. It run as normal

`ros2 launch lidarslam lidarslam.launch.py`

[image]

Next, I run RVIZ2 to visualise the path and modified_path. Also, have no issue.

[image]

[image]

After running lidarslam and rviz2, I play the ros2 bag. The way I play the ros bag is as following:

`ros2 bag play file.bag`

[image]

After few seconds, the vehicle is moving and now we can see the path. 

[image]

Suddenly, there is an error occured in lidarslam. The error is as following:

```
[scanmatcher_node-1] [pcl::VoxelGrid::applyFilter] Leaf size is too small for the input dataset. Integer indices would overflow.
[ERROR] [scanmatcher_node-1]: process has died [pid 121510, exit code -8, cmd '/home/reka/ros2_ws/install/scanmatcher/lib/scanmatcher/scanmatcher_node --ros-args --params-file /home/reka/ros2_ws/install/lidarslam/share/lidarslam/param/lidarslam.yaml -r /input_cloud:=/c32/lslidar_point_cloud'].
```

[image]

Now, the lidarslam is crash. and the path is not moving anymore. Can I know what the reason behind this and how to fix it?

I tried to fix it by increasing the leaf size in params.yaml, but seems not working.

Tester2009, 2023-05-30:
Thank you @rsasaki0109 for your fast response. I have enabled debug_flag by change it to true value.

Following is the video that I capture when running lidarslam.

[screencast.webm](https://github.com/rsasaki0109/lidarslam_ros2/assets/3969897/a14ebe17-dd1a-459b-8087-6cda8ad9f083)

rsasaki0109, 2023-05-30:
@Tester2009 I would like you to provide me with the YAML file.

Tester2009, 2023-05-30:
@rsasaki0109 Here is my YAML file.

```
scan_matcher:
  ros__parameters:
    global_frame_id: "map"
    robot_frame_id: "base_link"
    registration_method: "NDT"
    ndt_resolution: 2.0
    ndt_num_threads: 2
    gicp_corr_dist_threshold: 5.0
    trans_for_mapupdate: 1.5
    vg_size_for_input: 0.5
    vg_size_for_map: 0.1
    use_min_max_filter: true
    scan_min_range: 1.0
    scan_max_range: 200.0
    scan_period: 0.2
    map_publish_period: 15.0
    num_targeted_cloud: 20
    set_initial_pose: true
    initial_pose_x: 0.0
    initial_pose_y: 0.0
    initial_pose_z: 0.0
    initial_pose_qx: 0.0
    initial_pose_qy: 0.0
    initial_pose_qz: 0.0
    initial_pose_qw: 1.0
    use_imu: false
    use_odom: false
    debug_flag: true

graph_based_slam:
    ros__parameters:
      registration_method: "NDT"
      ndt_resolution: 1.0
      ndt_num_threads: 2
      voxel_leaf_size: 0.1
      loop_detection_period: 3000
      threshold_loop_closure_score: 0.7
      distance_loop_closure: 100.0
      range_of_searching_loop_closure: 20.0
      search_submap_num: 2
      num_adjacent_pose_cnstraints: 5
      use_save_map_in_loop: true
      debug_flag: true
```

rsasaki0109, 2023-05-30:
Hmmm... I'm a little unsure.
I think I need to know the outline of the filtered scan point cloud and the number of point clouds when the error occurs.
I'm a bit busy so I can't get started right away, but if you provide a rosbag I can take a look at it in my spare time.

Tester2009, 2023-05-31:
Hi @rsasaki0109 . Here is the link to the rosbag. The size of rosbag is 2GB compressed.

https://drive.google.com/file/d/1c8Aq0SliIqB5iGAl35CViehIKL8L74Zz/view?usp=share_link

Look forward to your reply soon!

Tester2009, 2023-06-06:
Hi @rsasaki0109 good day! 

May I know if there any update on this issue?

rsasaki0109, 2023-06-06:
I apologize, it slipped my mind. At a glance, it seems that the voxel grid filter inside updateMap, which is running in parallel, might be the issue.
https://github.com/rsasaki0109/lidarslam_ros2/blob/humble/scanmatcher/src/scanmatcher_component.cpp#L422

Due to time constraints, I'll check again on a weekday evening this week.

colconbator, 2023-07-14:
I am facing the same problem, whenever I play rosbag2 play bag file command the bag is getting closed and rviz is not showing any map

rsasaki0109, 2023-12-25:
memo
https://github.com/autowarefoundation/autoware.universe/pull/5951

sbrodeur, 2024-08-15:
I had the same issue for my custom recordings with an OS1-64. I just needed to adjust the parameters vg_size_for_input and vg_size_for_map. In my case, increasing vg_size_for_map to 0.5 solved the crash.

miku54, 2024-11-15:
> I had the same issue for my custom recordings with an OS1-64. I just needed to adjust the parameters vg_size_for_input and vg_size_for_map. In my case, increasing vg_size_for_map to 0.5 solved the crash.

I also encountered the same problem during debugging. Which parameter is vg_size_for_map? I can't find it.

sbrodeur, 2024-11-15:
The parameter is documented here: https://github.com/rsasaki0109/lidarslam_ros2/blob/develop/README.md?plain=1#L85
You can set the value in the ROS parameter file that is configuring the scan matching node, as in here: https://github.com/rsasaki0109/lidarslam_ros2/blob/develop/lidarslam/param/lidarslam.yaml#L11

rsasaki0109, 2024-11-15:
If it crashes, it's better to increase vg_size_for_map, but in terms of accuracy, it's best to keep it as small as possible.