rsasaki0109/lidar_slam_ros2 #71: Intel Realse Sense on the manipulator

delipl, 2023-06-09:
Hi @rsasaki0109!
Is it possible to use your 3D mapping with the mobile manipulator with the 3D depth camera on it?
[Screencast from 09.06.2023 21:17:49.webm](https://github.com/rsasaki0109/lidarslam_ros2/assets/33808634/f7126174-2eb5-4098-82d5-6765f1e1418c)
The `base_link` is the base of the platform. For this video the odom is static. The platform is not moving.

rsasaki0109, 2023-06-11:
You don't need to worry about the 'Failed to find match for field 'intensity'' message. 

RealSense has a narrower field of view and less precision compared to rotating lidar, making scan matching more challenging. 

If you're measuring indoors, you may want to adjust the 'ndt_resolution' parameter to 1 and the 'trans_for_mapupdate' parameter to around 0.5 to 1.0.

rsasaki0109, 2024-01-19:
As there has been no activity for a long time, I'm closing this issue. It can be reopened if necessary.
For example, I can review it if a rosbag is provided.

closed 2024-01-19 by rsasaki0109