rsasaki0109/lidar_slam_ros2 #72: /map topic will not be displayed properly in rviz2

Yoshihito-M, 2023-07-28:
Hello rsasaki0109 

I'm using the lidarslam_ros2 with Livox MID-360, but /map topic will not be displayed properly in rviz2.
I tried to modify some parameters of _lidarslam.launch.py_ in order to use MID-360, but I couldn't get a good result.
The modified parameters are shown as follows in bold font:

`    mapping = launch_ros.actions.Node(
        package='scanmatcher',
        executable='scanmatcher_node',
        parameters=[main_param_dir],
        **remappings=[('/input_cloud','/livox/lidar')],**
        output='screen'
        )

    tf = launch_ros.actions.Node(
        package='tf2_ros',
        executable='static_transform_publisher',
        **arguments=['0','0','0','0','0','0','1','base_link','livox_frame']**
        )`.

I would like you to give me some advice to solve the above issue.

Best regards

rsasaki0109, 2023-07-28:
Could you please run `ros2 topic info /livox/lidar -v` to check if the rosbag player and scan matcher are properly connected?

Firstly, are you able to estimate the trajectory using scan matching? You may want to set the `debug_flag` to true in `lidarslam.yaml` and examine the terminal output or check `/path`.

If `/path` appears but the trajectory seems chaotic, adjusting parameters such as ndt resolution or voxel grid size might improve the results. For more information, consider searching past issues using the parameter names.

Yoshihito-M, 2023-07-28:
＠rsasaki0109

Thank you for your quick response.

After waiting for about 1 minute, /map topic is shown in Rviz2.
Is my understanding correct that /path topic is shown more quickly than /map topic？

rsasaki0109, 2023-07-28:
Yes, that is correct. The path is shown more quickly. It takes some time from the estimation before the map is displayed.

Yoshihito-M, 2023-07-31:
＠rsasaki0109

I understood.
Thank you again for your help.

closed 2023-07-31 by Yoshihito-M