rsasaki0109/lidar_slam_ros2 #74: Build Issue (PCL version error)

NadimDeeb, 2023-08-28:
I am using ROS Foxy.
I've followed the steps described for cloning the repository in the readme.md file but when I `colcon build` my workspace i always get the following issue:
`--- stderr: ndt_omp_ros2                                                                         
CMake Error at CMakeLists.txt:23 (find_package):
  Could not find a configuration file for package "PCL" that is compatible
  with requested version "1.12".

  The following configuration files were considered but not accepted:

    /usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake, version: 1.10.0
    /lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake, version: 1.10.0

make: *** [Makefile:370: cmake_check_build_system] Error 1
---
`
I tried to upgrade my PCL version to 1.12 but it doesn't seem to do the trick.
I tried to download the repository as a zip and then run the `git submodule update --init --recursive` inside the package but it just gave me a longer series of g2o errors:
`--- stderr: graph_based_slam                                                       
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** io features related to libusb-1.0 will be disabled
CMake Warning (dev) at /usr/share/cmake-3.16/Modules/FindOpenGL.cmake:275 (message):
  Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
  available.  Run "cmake --help-policy CMP0072" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  FindOpenGL found both a legacy GL library:

    OPENGL_gl_LIBRARY: /usr/lib/x86_64-linux-gnu/libGL.so

  and GLVND libraries for OpenGL and GLX:

    OPENGL_opengl_LIBRARY: /usr/lib/x86_64-linux-gnu/libOpenGL.so
    OPENGL_glx_LIBRARY: /usr/lib/x86_64-linux-gnu/libGLX.so

  OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
  compatibility with CMake 3.10 and below the legacy GL library will be used.
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/CMakeFindDependencyMacro.cmake:47 (find_package)
  /usr/local/lib/cmake/g2o/g2oConfig.cmake:4 (find_dependency)
  CMakeLists.txt:33 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

In file included from /usr/local/include/g2o/core/sparse_block_matrix.h:43,
                 from /usr/local/include/g2o/core/sparse_optimizer.h:34,
                 from /home/nadimdib/ros2_3D/src/lidarslam_ros2/graph_based_slam/include/graph_based_slam/graph_based_slam_component.h:75,
                 from /home/nadimdib/ros2_3D/src/lidarslam_ros2/graph_based_slam/src/graph_based_slam_component.cpp:1:
/usr/local/include/g2o/core/matrix_structure.h:57:19: error: ‘std::string_view’ has not been declared
   57 |   bool write(std::string_view filename) const;
      |                   ^~~~~~~~~~~
In file included from /usr/local/include/g2o/core/base_fixed_sized_edge.h:39,
                 from /usr/local/include/g2o/core/base_binary_edge.h:30,
                 from /usr/local/include/g2o/types/slam3d/edge_se3.h:30,
                 from /home/nadimdib/ros2_3D/src/lidarslam_ros2/graph_based_slam/include/graph_based_slam/graph_based_slam_component.h:83,
                 from /home/nadimdib/ros2_3D/src/lidarslam_ros2/graph_based_slam/src/graph_based_slam_component.cpp:1:
/usr/local/include/g2o/stuff/tuple_tools.h: In function ‘void g2o::tuple_apply_i(F&&, T&, int)’:
/usr/local/include/g2o/stuff/tuple_tools.h:41:46: error: ‘tuple_size_v’ is not a member of ‘std’; did you mean ‘tuple_size’?
   41 |       f, t, i, std::make_index_sequence<std::tuple_size_v<std::decay_t<T>>>());
      |                                              ^~~~~~~~~~~~
      |                                              tuple_size
/usr/local/include/g2o/stuff/tuple_tools.h:41:46: error: ‘tuple_size_v’ is not a member of ‘std’; did you mean ‘tuple_size’?
   41 |       f, t, i, std::make_index_sequence<std::tuple_size_v<std::decay_t<T>>>());
      |                                              ^~~~~~~~~~~~
      |                                              tuple_size
/usr/local/include/g2o/stuff/tuple_tools.h:41:73: error: template argument 1 is invalid
   41 |       f, t, i, std::make_index_sequence<std::tuple_size_v<std::decay_t<T>>>());
      |                                                                         ^~
/usr/local/include/g2o/stuff/tuple_tools.h:41:77: error: expected primary-expression before ‘)’ token
   41 |       f, t, i, std::make_index_sequence<std::tuple_size_v<std::decay_t<T>>>());
      |                                                                             ^
/home/nadimdib/ros2_3D/src/lidarslam_ros2/graph_based_slam/src/graph_based_slam_component.cpp: In member function ‘void graphslam::GraphBasedSlamComponent::doPoseAdjustment(lidarslam_msgs::msg::MapArray, bool)’:
/home/nadimdib/ros2_3D/src/lidarslam_ros2/graph_based_slam/src/graph_based_slam_component.cpp:272:10: error: ‘make_unique’ is not a member of ‘g2o’; did you mean ‘std::make_unique’?
  272 |     g2o::make_unique<g2o::LinearSolverEigen<g2o::BlockSolver_6_3::PoseMatrixType>>();
      |          ^~~~~~~~~~~
In file included from /usr/include/c++/9/memory:80,
                 from /opt/ros/foxy/include/rclcpp/rclcpp.hpp:144,
                 from /home/nadimdib/ros2_3D/src/lidarslam_ros2/graph_based_slam/include/graph_based_slam/graph_based_slam_component.h:42,
                 from /home/nadimdib/ros2_3D/src/lidarslam_ros2/graph_based_slam/src/graph_based_slam_component.cpp:1:
/usr/include/c++/9/bits/unique_ptr.h:868:5: note: ‘std::make_unique’ declared here
  868 |     make_unique(_Args&&...) = delete;
      |     ^~~~~~~~~~~
/home/nadimdib/ros2_3D/src/lidarslam_ros2/graph_based_slam/src/graph_based_slam_component.cpp:272:81: error: expected primary-expression before ‘>’ token
  272 |     g2o::make_unique<g2o::LinearSolverEigen<g2o::BlockSolver_6_3::PoseMatrixType>>();
      |                                                                                 ^~
/home/nadimdib/ros2_3D/src/lidarslam_ros2/graph_based_slam/src/graph_based_slam_component.cpp:272:84: error: expected primary-expression before ‘)’ token
  272 |     g2o::make_unique<g2o::LinearSolverEigen<g2o::BlockSolver_6_3::PoseMatrixType>>();
      |                                                                                    ^
/home/nadimdib/ros2_3D/src/lidarslam_ros2/graph_based_slam/src/graph_based_slam_component.cpp:274:10: error: ‘make_unique’ is not a member of ‘g2o’; did you mean ‘std::make_unique’?
  274 |     g2o::make_unique<g2o::BlockSolver_6_3>(std::move(linear_solver)));
      |          ^~~~~~~~~~~
In file included from /usr/include/c++/9/memory:80,
                 from /opt/ros/foxy/include/rclcpp/rclcpp.hpp:144,
                 from /home/nadimdib/ros2_3D/src/lidarslam_ros2/graph_based_slam/include/graph_based_slam/graph_based_slam_component.h:42,
                 from /home/nadimdib/ros2_3D/src/lidarslam_ros2/graph_based_slam/src/graph_based_slam_component.cpp:1:
/usr/include/c++/9/bits/unique_ptr.h:868:5: note: ‘std::make_unique’ declared here
  868 |     make_unique(_Args&&...) = delete;
      |     ^~~~~~~~~~~
/home/nadimdib/ros2_3D/src/lidarslam_ros2/graph_based_slam/src/graph_based_slam_component.cpp:274:42: error: expected primary-expression before ‘>’ token
  274 |     g2o::make_unique<g2o::BlockSolver_6_3>(std::move(linear_solver)));
      |                                          ^
make[2]: *** [CMakeFiles/graph_based_slam_component.dir/build.make:63: CMakeFiles/graph_based_slam_component.dir/src/graph_based_slam_component.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:107: CMakeFiles/graph_based_slam_component.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---`

rsasaki0109, 2023-08-28:
Did you use the humble branch? Please use the foxy branch.
https://github.com/rsasaki0109/lidarslam_ros2/tree/foxy

NadimDeeb, 2023-08-29:
When I git clone, I am obligated to use the the general branch correct? 
So I have to run `git clone --recursive https://github.com/rsasaki0109/lidarslam_ros2` and then run `cd lidarslam_ros2 && git checkout foxy` followed by building the project from the workspace directory using `colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release`

rsasaki0109, 2023-08-29:
With that, the submodule will use the one from the 'humble' branch. Please use the following command.

```
git clone --recursive https://github.com/rsasaki0109/lidarslam_ros2 -b foxy
```

NadimDeeb, 2023-08-29:
I ran the git clone command provided and then built the workspace like instructed and I got a really long error for a build failure.
Here are some snipets (since the error message can't be fully read in terminal):
`/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h: In instantiation of ‘void Eigen::internal::resize_if_allowed(DstXprType&, const SrcXprType&, const Eigen::internal::assign_op<T1, T2>&) [with DstXprType = Eigen::Matrix<int, 4, 1>; SrcXprType = Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<int>, Eigen::Matrix<int, 4, 1> >; T1 = int; T2 = int]’:
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:734:20:   required from ‘void Eigen::internal::call_dense_assignment_loop(DstXprType&, const SrcXprType&, const Functor&) [with DstXprType = Eigen::Matrix<int, 4, 1>; SrcXprType = Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<int>, Eigen::Matrix<int, 4, 1> >; Functor = Eigen::internal::assign_op<int, int>]’
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:879:31:   required from ‘static void Eigen::internal::Assignment<DstXprType, SrcXprType, Functor, Eigen::internal::Dense2Dense, Weak>::run(DstXprType&, const SrcXprType&, const Functor&) [with DstXprType = Eigen::Matrix<int, 4, 1>; SrcXprType = Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<int>, Eigen::Matrix<int, 4, 1> >; Functor = Eigen::internal::assign_op<int, int>; Weak = void]’
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:836:49:   required from ‘void Eigen::internal::call_assignment_no_alias(Dst&, const Src&, const Func&) [with Dst = Eigen::Matrix<int, 4, 1>; Src = Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<int>, Eigen::Matrix<int, 4, 1> >; Func = Eigen::internal::assign_op<int, int>]’
/usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:732:41:   required from ‘Derived& Eigen::PlainObjectBase<Derived>::_set_noalias(const Eigen::DenseBase<OtherDerived>&) [with OtherDerived = Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<int>, Eigen::Matrix<int, 4, 1> >; Derived = Eigen::Matrix<int, 4, 1>]’
/usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:816:7:   required from ‘void Eigen::PlainObjectBase<Derived>::_init1(const Eigen::DenseBase<ElseDerived>&) [with T = Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<int>, Eigen::Matrix<int, 4, 1> >; OtherDerived = Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<int>, Eigen::Matrix<int, 4, 1> >; Derived = Eigen::Matrix<int, 4, 1>]’
/usr/include/eigen3/Eigen/src/Core/Matrix.h:294:31:   required from ‘Eigen::Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols>::Matrix(const T&) [with T = Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<int>, Eigen::Matrix<int, 4, 1> >; _Scalar = int; int _Rows = 4; int _Cols = 1; int _Options = 0; int _MaxRows = 4; int _MaxCols = 1]’
/usr/local/include/pcl-1.12/pcl/filters/voxel_grid.h:533:33:   required from here
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:719:12: error: ‘class Eigen::Matrix<int, 4, 1>’ has no member named ‘rows’; did you mean ‘row’?
  719 |   if(((dst.rows()!=dstRows) || (dst.cols()!=dstCols)))
      |        ~~~~^~~~
      |        row
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:719:37: error: ‘class Eigen::Matrix<int, 4, 1>’ has no member named ‘cols’; did you mean ‘cos’?
  719 |   if(((dst.rows()!=dstRows) || (dst.cols()!=dstCols)))
      |                                 ~~~~^~~~
      |                                 cos
`
`In file included from /usr/include/eigen3/Eigen/Core:440,
                 from /usr/local/include/pcl-1.12/pcl/memory.h:48,
                 from /usr/local/include/pcl-1.12/pcl/pcl_base.h:46,
                 from /usr/local/include/pcl-1.12/pcl/filters/filter.h:42,
                 from /usr/local/include/pcl-1.12/pcl/filters/voxel_grid.h:42,
                 from /home/nadimdib/Test/src/lidarslam_ros2/Thirdparty/ndt_omp_ros2/include/pclomp/voxel_grid_covariance_omp.h:42,
                 from /home/nadimdib/Test/src/lidarslam_ros2/Thirdparty/ndt_omp_ros2/src/pclomp/voxel_grid_covariance_omp.cpp:1:
/usr/include/eigen3/Eigen/src/Core/CoreEvaluators.h: In instantiation of ‘Eigen::internal::evaluator_wrapper_base<XprType>::evaluator_wrapper_base(const ArgType&) [with XprType = Eigen::ArrayWrapper<Eigen::Matrix<int, 4, 1> >; Eigen::internal::evaluator_wrapper_base<XprType>::ArgType = Eigen::Matrix<int, 4, 1>]’:
/usr/include/eigen3/Eigen/src/Core/CoreEvaluators.h:1422:81:   required from ‘Eigen::internal::unary_evaluator<Eigen::ArrayWrapper<Derived> >::unary_evaluator(const XprType&) [with TArgType = Eigen::Matrix<int, 4, 1>; typename Eigen::ArrayWrapper<Derived>::Scalar = int; typename Eigen::internal::evaluator_traits<typename Eigen::ArrayWrapper<Derived>::NestedExpression>::Kind = Eigen::internal::IndexBased; Eigen::internal::unary_evaluator<Eigen::ArrayWrapper<Derived> >::XprType = Eigen::ArrayWrapper<Eigen::Matrix<int, 4, 1> >]’
/usr/include/eigen3/Eigen/src/Core/CoreEvaluators.h:93:64:   required from ‘Eigen::internal::evaluator<T>::evaluator(const T&) [with T = Eigen::ArrayWrapper<Eigen::Matrix<int, 4, 1> >]’
/usr/include/eigen3/Eigen/src/Core/CoreEvaluators.h:103:54:   required from ‘Eigen::internal::evaluator<const T>::evaluator(const T&) [with T = Eigen::ArrayWrapper<Eigen::Matrix<int, 4, 1> >]’
[... cut ...]
  374 | xpr.data()+xpr.innerStride()*(XprTypeIsRowMajor?startCol:startRow) + xpr.outerStride()*(XprTypeIsRowMajor?startRow:startCol), blockRows, blockCols),

/usr/include/eigen3/Eigen/src/Core/Block.h:375:62: error: using invalid field ‘Eigen::internal::BlockImpl_dense<XprType, BlockRows, BlockCols, InnerPanel, true>::m_startRow’
  375 |         m_xpr(xpr), m_startRow(startRow), m_startCol(startCol)
      |                                                              ^
/usr/include/eigen3/Eigen/src/Core/Block.h:375:62: error: using invalid field ‘Eigen::internal::BlockImpl_dense<XprType, BlockRows, BlockCols, InnerPanel, true>::m_startCol’
`

rsasaki0109, 2023-08-29:
How about deleting the 'install' and 'build' from the workspace and then building it again?

rsasaki0109, 2023-08-29:
Do you have a different version of PCL installed than the default one in ROS2 Foxy? If so, you might need to uninstall it.

rsasaki0109, 2023-08-29:
I haven't used ros2 foxy for a while, so I forgot the pcl version.

NadimDeeb, 2023-08-29:
I ended up running the `locale pcl` and looking through any possible versions and found out I have pcl-1.10 and pcl-1.12 installed as well as the pcl debian package. I ended up calling `sudo rm -r ` on the paths to these location leaving only the pcl debian package which made the error no longer pop up.
I am now getting a g2o related error with this following erro log, do you have any idea why:
`Starting >>> lidarslam_msgs
Starting >>> ndt_omp_ros2
Finished <<< lidarslam_msgs [4.04s]                                                                 
--- stderr: ndt_omp_ros2                                
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** io features related to libusb-1.0 will be disabled
---
Finished <<< ndt_omp_ros2 [31.5s]
Starting >>> graph_based_slam
Starting >>> scanmatcher
--- stderr: graph_based_slam                                                                           
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** io features related to libusb-1.0 will be disabled
CMake Warning (dev) at /usr/share/cmake-3.16/Modules/FindOpenGL.cmake:275 (message):
  Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
  available.  Run "cmake --help-policy CMP0072" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  FindOpenGL found both a legacy GL library:

    OPENGL_gl_LIBRARY: /usr/lib/x86_64-linux-gnu/libGL.so

  and GLVND libraries for OpenGL and GLX:

    OPENGL_opengl_LIBRARY: /usr/lib/x86_64-linux-gnu/libOpenGL.so
    OPENGL_glx_LIBRARY: /usr/lib/x86_64-linux-gnu/libGLX.so

  OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
  compatibility with CMake 3.10 and below the legacy GL library will be used.
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/CMakeFindDependencyMacro.cmake:47 (find_package)
  /usr/local/lib/cmake/g2o/g2oConfig.cmake:4 (find_dependency)
  CMakeLists.txt:33 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

In file included from /usr/local/include/g2o/core/sparse_block_matrix.h:43,
                 from /usr/local/include/g2o/core/sparse_optimizer.h:34,
                 from /home/nadimdib/test/src/lidarslam_ros2/graph_based_slam/include/graph_based_slam/graph_based_slam_component.h:75,
                 from /home/nadimdib/test/src/lidarslam_ros2/graph_based_slam/src/graph_based_slam_component.cpp:1:
/usr/local/include/g2o/core/matrix_structure.h:57:19: error: ‘std::string_view’ has not been declared
   57 |   bool write(std::string_view filename) const;
      |                   ^~~~~~~~~~~
In file included from /usr/local/include/g2o/core/base_fixed_sized_edge.h:39,
                 from /usr/local/include/g2o/core/base_binary_edge.h:30,
                 from /usr/local/include/g2o/types/slam3d/edge_se3.h:30,
                 from /home/nadimdib/test/src/lidarslam_ros2/graph_based_slam/include/graph_based_slam/graph_based_slam_component.h:83,
                 from /home/nadimdib/test/src/lidarslam_ros2/graph_based_slam/src/graph_based_slam_component.cpp:1:
/usr/local/include/g2o/stuff/tuple_tools.h: In function ‘void g2o::tuple_apply_i(F&&, T&, int)’:
/usr/local/include/g2o/stuff/tuple_tools.h:41:46: error: ‘tuple_size_v’ is not a member of ‘std’; did you mean ‘tuple_size’?
   41 |       f, t, i, std::make_index_sequence<std::tuple_size_v<std::decay_t<T>>>());
      |                                              ^~~~~~~~~~~~
      |                                              tuple_size
/usr/local/include/g2o/stuff/tuple_tools.h:41:46: error: ‘tuple_size_v’ is not a member of ‘std’; did you mean ‘tuple_size’?
   41 |       f, t, i, std::make_index_sequence<std::tuple_size_v<std::decay_t<T>>>());
      |                                              ^~~~~~~~~~~~
      |                                              tuple_size
/usr/local/include/g2o/stuff/tuple_tools.h:41:73: error: template argument 1 is invalid
   41 |       f, t, i, std::make_index_sequence<std::tuple_size_v<std::decay_t<T>>>());
      |                                                                         ^~
/usr/local/include/g2o/stuff/tuple_tools.h:41:77: error: expected primary-expression before ‘)’ token
   41 |       f, t, i, std::make_index_sequence<std::tuple_size_v<std::decay_t<T>>>());
      |                                                                             ^
/home/nadimdib/test/src/lidarslam_ros2/graph_based_slam/src/graph_based_slam_component.cpp: In member function ‘void graphslam::GraphBasedSlamComponent::doPoseAdjustment(lidarslam_msgs::msg::MapArray, bool)’:
/home/nadimdib/test/src/lidarslam_ros2/graph_based_slam/src/graph_based_slam_component.cpp:272:10: error: ‘make_unique’ is not a member of ‘g2o’; did you mean ‘std::make_unique’?
  272 |     g2o::make_unique<g2o::LinearSolverEigen<g2o::BlockSolver_6_3::PoseMatrixType>>();
      |          ^~~~~~~~~~~
[... 13 lines cut ...]
/home/nadimdib/test/src/lidarslam_ros2/graph_based_slam/src/graph_based_slam_component.cpp:274:10: error: ‘make_unique’ is not a member of ‘g2o’; did you mean ‘std::make_unique’?
  274 |     g2o::make_unique<g2o::BlockSolver_6_3>(std::move(linear_solver)));
      |          ^~~~~~~~~~~
In file included from /usr/include/c++/9/memory:80,
[... cut ...]
make[2]: *** [CMakeFiles/graph_based_slam_component.dir/build.make:63: CMakeFiles/graph_based_slam_component.dir/src/graph_based_slam_component.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:107: CMakeFiles/graph_based_slam_component.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< graph_based_slam [21.6s, exited with code 2]
Aborted  <<< scanmatcher [1min 20s]                                         

Summary: 2 packages finished [1min 52s]
  1 package failed: graph_based_slam
  1 package aborted: scanmatcher
  3 packages had stderr output: graph_based_slam ndt_omp_ros2 scanmatcher
  1 package not processed`

NadimDeeb, 2023-08-30:
Hey again, I noticed you labeled the foxy branch as failing. I managed to get it to build.
-First I changed the C++ version from 14 to 17 which resolved a bunch of errors related to the compiler.
-Second I changed the following namespaces form g2o to std (but I don't think that's a valid solution) at line 269 to 276:
      ```
g2o::SparseOptimizer optimizer;
  optimizer.setVerbose(false);
  std::unique_ptr<g2o::BlockSolver_6_3::LinearSolverType> linear_solver =
    // g2o::make_unique<g2o::LinearSolverEigen<g2o::BlockSolver_6_3::PoseMatrixType>>();
    std::make_unique<g2o::LinearSolverEigen<g2o::BlockSolver_6_3::PoseMatrixType>>();
  g2o::OptimizationAlgorithmLevenberg * solver = new g2o::OptimizationAlgorithmLevenberg(
    // g2o::make_unique<g2o::BlockSolver_6_3>(std::move(linear_solver)));
    std::make_unique<g2o::BlockSolver_6_3>(std::move(linear_solver)));
```

rsasaki0109, 2024-01-19:
It seems that the issue has been resolved, so I am closing it.

closed 2024-01-19 by rsasaki0109