rsasaki0109/lidar_slam_ros2 #78: Frame of map

lv-alerion, 2023-09-20:
Hi,

thank you for your repo !
I have some trouble when launching `lidarslam`: 

- the /map pointcloud is in an " " frame rather than the map frame (that exists).
  I guess this comes from the scanmatcher but cannot figure out how to correct it.
  `/path` is working correctly
  
- I get no output on `/modified_*` using debug mode on graphbasedslam does not help

The last message I get from logs is from the scan_matcher: `create a first map`.
I get a single map from it

Do you have any idea about what I am doing wrong ?

Here is the param file I am using :

```
scan_matcher:
  ros__parameters:
    global_frame_id: "map"
    robot_frame_id: "myrobot"
    registration_method: "NDT"
    ndt_resolution: 5.0
    ndt_num_threads: 0
    gicp_corr_dist_threshold: 5.0
    trans_for_mapupdate: 1.5
    vg_size_for_input: 0.2
    vg_size_for_map: 0.05
    use_min_max_filter: true
    scan_min_range: 1.0
    scan_max_range: 100.0
    scan_period: 0.1
    map_publish_period: 15.0
    num_targeted_cloud: 10
    set_initial_pose: true
    initial_pose_x: 0.0
    initial_pose_y: 0.0
    initial_pose_z: 0.0
    initial_pose_qx: 0.0
    initial_pose_qy: 0.0
    initial_pose_qz: 0.0
    initial_pose_qw: 1.0
    publish_tf: true
    use_imu: false
    use_odom: false
    debug_flag: false

graph_based_slam:
    ros__parameters:
      registration_method: "NDT"
      ndt_resolution: 5.0
      ndt_num_threads: 0
      voxel_leaf_size: 0.2
      loop_detection_period: 3000
      threshold_loop_closure_score: 0.7
      distance_loop_closure: 100.0
      range_of_searching_loop_closure: 20.0
      search_submap_num: 2
      num_adjacent_pose_cnstraints: 5
      use_save_map_in_loop: true
      debug_flag: false
``` 

Modified launch file
```python

def generate_launch_description():

    main_param_dir = launch.substitutions.LaunchConfiguration(
        "main_param_dir",
        default=os.path.join(
            get_package_share_directory("lidarslam"),
            "param",
            "lidarslam-myrobot.yml",
        ),
    )

    mapping = launch_ros.actions.Node(
        package="scanmatcher",
        executable="scanmatcher_node",
        parameters=[main_param_dir],
        remappings=[("/input_cloud", "/ouster/points")],
        output="screen",
    )

    # tf = launch_ros.actions.Node(
    #     package="tf2_ros",
    #     executable="static_transform_publisher",
    #     arguments=[
    #         "0",
    #         "0",
    #         "0",
    #         "0",
    #         "0",
    #         "0",
    #         "1",
    #         "myrobot",
    #         "os_sensor",
    #     ],
    # )

    graphbasedslam = launch_ros.actions.Node(
        package="graph_based_slam",
        executable="graph_based_slam_node",
        parameters=[main_param_dir],
        output="screen",
    )

    return launch.LaunchDescription(
        [
            launch.actions.DeclareLaunchArgument(
                "main_param_dir",
                default_value=main_param_dir,
                description="Full path to main parameter file to load",
            ),
            mapping,
            # tf,
            graphbasedslam,
        ]
    )
```

rsasaki0109, 2023-09-25:
I didn't receive a notification, so my reply was delayed.
That's a strange issue. Can you share the terminal output when you boot up?

lv-alerion, 2023-09-25:
Hi,
here is the requested output.
Could it be related to libg2o ? I have been thinking it is only for optimization

```
shuttle-lidarslam-1  | sourcing   /opt/ros/humble/install/setup.bash
shuttle-lidarslam-1  | not found: "/opt/ros/humble/install/share/libg2o/local_setup.bash"
shuttle-lidarslam-1  | ROS_DISTRO humble
shuttle-lidarslam-1  | ROS_ROOT   /opt/ros/humble
shuttle-lidarslam-1  | [INFO] [launch]: All log files can be found below /root/.ros/log/2023-09-20-13-12-15-102974-shuttle-1-1
shuttle-lidarslam-1  | [INFO] [launch]: Default logging verbosity is set to INFO
shuttle-lidarslam-1  | [INFO] [scanmatcher_node-1]: process started with pid [56]
shuttle-lidarslam-1  | [INFO] [graph_based_slam_node-2]: process started with pid [58]
shuttle-lidarslam-1  | [scanmatcher_node-1] registration_method:NDT
shuttle-lidarslam-1  | [scanmatcher_node-1] ndt_resolution[m]:5
shuttle-lidarslam-1  | [scanmatcher_node-1] ndt_num_threads:0
shuttle-lidarslam-1  | [scanmatcher_node-1] gicp_corr_dist_threshold[m]:5
shuttle-lidarslam-1  | [scanmatcher_node-1] trans_for_mapupdate[m]:1.5
shuttle-lidarslam-1  | [scanmatcher_node-1] vg_size_for_input[m]:0.2
shuttle-lidarslam-1  | [scanmatcher_node-1] vg_size_for_map[m]:0.05
shuttle-lidarslam-1  | [scanmatcher_node-1] use_min_max_filter:true
shuttle-lidarslam-1  | [scanmatcher_node-1] scan_min_range[m]:1
shuttle-lidarslam-1  | [scanmatcher_node-1] scan_max_range[m]:100
shuttle-lidarslam-1  | [scanmatcher_node-1] set_initial_pose:true
shuttle-lidarslam-1  | [scanmatcher_node-1] publish_tf:true
shuttle-lidarslam-1  | [scanmatcher_node-1] use_odom:false
shuttle-lidarslam-1  | [scanmatcher_node-1] use_imu:false
shuttle-lidarslam-1  | [scanmatcher_node-1] scan_period[sec]:0.1
shuttle-lidarslam-1  | [scanmatcher_node-1] debug_flag:false
shuttle-lidarslam-1  | [scanmatcher_node-1] map_publish_period[sec]:15
shuttle-lidarslam-1  | [scanmatcher_node-1] num_targeted_cloud:10
shuttle-lidarslam-1  | [scanmatcher_node-1] ------------------
shuttle-lidarslam-1  | [scanmatcher_node-1] [INFO] [1695215535.471805656] [scan_matcher]: initialize Publishers and Subscribers
shuttle-lidarslam-1  | [scanmatcher_node-1] [INFO] [1695215535.476068351] [scan_matcher]: initialization end
shuttle-lidarslam-1  | [graph_based_slam_node-2] [INFO] [1695215535.666181254] [graph_based_slam]: initialization start
shuttle-lidarslam-1  | [graph_based_slam_node-2] registration_method:NDT
shuttle-lidarslam-1  | [graph_based_slam_node-2] voxel_leaf_size[m]:0.2
shuttle-lidarslam-1  | [graph_based_slam_node-2] ndt_resolution[m]:5
shuttle-lidarslam-1  | [graph_based_slam_node-2] ndt_num_threads:0
shuttle-lidarslam-1  | [graph_based_slam_node-2] loop_detection_period[Hz]:3000
shuttle-lidarslam-1  | [graph_based_slam_node-2] threshold_loop_closure_score:0.7
shuttle-lidarslam-1  | [graph_based_slam_node-2] distance_loop_closure[m]:100
shuttle-lidarslam-1  | [graph_based_slam_node-2] range_of_searching_loop_closure[m]:20
shuttle-lidarslam-1  | [graph_based_slam_node-2] search_submap_num:2
shuttle-lidarslam-1  | [graph_based_slam_node-2] num_adjacent_pose_cnstraints:5
shuttle-lidarslam-1  | [graph_based_slam_node-2] use_save_map_in_loop:true
shuttle-lidarslam-1  | [graph_based_slam_node-2] debug_flag:true
shuttle-lidarslam-1  | [graph_based_slam_node-2] ------------------
shuttle-lidarslam-1  | [graph_based_slam_node-2] [INFO] [1695215535.669067734] [graph_based_slam]: initialize Publishers and Subscribers
shuttle-lidarslam-1  | [graph_based_slam_node-2] [INFO] [1695215535.676207693] [graph_based_slam]: initialization end
shuttle-lidarslam-1  | sourcing   /opt/ros/humble/install/setup.bash
shuttle-lidarslam-1  | not found: "/opt/ros/humble/install/share/libg2o/local_setup.bash"
shuttle-lidarslam-1  | ROS_DISTRO humble
shuttle-lidarslam-1  | ROS_ROOT   /opt/ros/humble
shuttle-lidarslam-1  | [INFO] [launch]: All log files can be found below /root/.ros/log/2023-09-25-12-22-11-106876-shuttle-1-1
shuttle-lidarslam-1  | [INFO] [launch]: Default logging verbosity is set to INFO
shuttle-lidarslam-1  | [INFO] [scanmatcher_node-1]: process started with pid [55]
shuttle-lidarslam-1  | [INFO] [graph_based_slam_node-2]: process started with pid [57]
shuttle-lidarslam-1  | [scanmatcher_node-1] registration_method:NDT
shuttle-lidarslam-1  | [scanmatcher_node-1] ndt_resolution[m]:5
shuttle-lidarslam-1  | [scanmatcher_node-1] ndt_num_threads:0
shuttle-lidarslam-1  | [scanmatcher_node-1] gicp_corr_dist_threshold[m]:5
shuttle-lidarslam-1  | [scanmatcher_node-1] trans_for_mapupdate[m]:1.5
shuttle-lidarslam-1  | [scanmatcher_node-1] vg_size_for_input[m]:0.2
shuttle-lidarslam-1  | [scanmatcher_node-1] vg_size_for_map[m]:0.05
shuttle-lidarslam-1  | [scanmatcher_node-1] use_min_max_filter:true
shuttle-lidarslam-1  | [scanmatcher_node-1] scan_min_range[m]:1
shuttle-lidarslam-1  | [scanmatcher_node-1] scan_max_range[m]:100
shuttle-lidarslam-1  | [scanmatcher_node-1] set_initial_pose:true
shuttle-lidarslam-1  | [scanmatcher_node-1] publish_tf:true
[... 51 lines cut ...]
shuttle-lidarslam-1  | [scanmatcher_node-1] map_publish_period[sec]:15
shuttle-lidarslam-1  | [scanmatcher_node-1] num_targeted_cloud:10
shuttle-lidarslam-1  | [scanmatcher_node-1] ------------------
shuttle-lidarslam-1  | [scanmatcher_node-1] [INFO] [1695644875.644985490] [scan_matcher]: initialize Publishers and Subscribers
shuttle-lidarslam-1  | [scanmatcher_node-1] [INFO] [1695644875.652884437] [scan_matcher]: initialization end
[... cut ...]
shuttle-lidarslam-1  | [graph_based_slam_node-2] [INFO] [1695644875.809387572] [graph_based_slam]: initialize Publishers and Subscribers
shuttle-lidarslam-1  | [graph_based_slam_node-2] [INFO] [1695644875.816277359] [graph_based_slam]: initialization end
shuttle-lidarslam-1  | [scanmatcher_node-1] [INFO] [1695644896.934588529] [scan_matcher]: create a first map
shuttle-lidarslam-1  | [graph_based_slam_node-2] searching Loop, num_submaps:2
shuttle-lidarslam-1  | [graph_based_slam_node-2] searching Loop, num_submaps:3
```

rsasaki0109, 2023-09-25:
Could you please check the value of the ‘frame_id’ in the ‘map’ using ‘ros2 param get’ or ‘std::cout’? It should be set to the ‘global_frame_id’.
https://github.com/rsasaki0109/lidarslam_ros2/blob/humble/scanmatcher/src/scanmatcher_component.cpp#L20

rsasaki0109, 2023-09-26:
It's strange that even though both 'path' and 'map' have the same frame_id, only one of them seems to be working correctly. I'm not sure about this. I'll contact you if I think of something else.

lv-alerion, 2023-09-26:
`ros2 param get /scan_matcher global_frame_id` outputs `map`, which does seem to me to be correct.

Ok, I have a fix: in my param file I had `robot_frame_id` set to `myrobot`. I thought this was ok because I am publishing a tf that links the `os_sensor` frame (ouster) to `myrobot` frame. Setting the `robot_frame_id` to `os_sensor` fixes it: it hinders a bit the frame hierarchy as instead of having

```
map
|- myrobot
    |- os_sensor
```
    
I get

```
map
|- os_sensor
my_robot
|- os_sensor
```

Thank you for your answers!

Now the map is produced, I am only missing the `modified_*` from graph-based-slam.

rsasaki0109, 2023-09-26:
> Now the map is produced, I am only missing the modified_* from graph-based-slam.

Either by performing loop closure or by calling ros2 service call /map_save std_srvs/Empty, the modified_* will be published.

lv-alerion, 2023-09-27:
I re-open this issue as I have met the frame problem again.

I try to change the namespace for the scanmatcher, which I have done uing the remappings argument

```python
remappings=[
("/input_cloud", "/my_ns/whatsneeded"),
("/map", "/my_ns/map"),
("/path", "/my_ns/path")
]
``` 

Result is exactly the same:
- a single map produced on a "" frame
- `/my_ns/path` working well
- no error triggered 

Trying to figure out what is happening, I suspect the `tf` performed by scanmatcher to be related to this.
When I set the `publish_tf` param to false, I do not get this strange  "" frame being produced. I even (sometimes ?) get the `/my_ns/map` published on the `map` frame, although I cannot really correlate it with the `/my_ns/path`.

However I would need to get the `tf` to link my robot to the map...

Hope these new elements help.

lv-alerion, 2023-09-27:
To be complete, here is some supplementary elements I have checked

Header of message on `/my_ns/map` when `publish_tf = false`
```
header:
  stamp:
    sec: 0
    nanosec: 0
  frame_id: map
```
Same, header of path
```
stamp:
  sec: [something plausible]
  nanosec: [something plausible]
frame_id: map
```

[Writing while testing]
And now I got it working with the `pulish_tf`... I could still see the first message published on `/my_ns/map` with the following header
```
header:
  stamp:
    sec: 0
    nanosec: 0
  frame_id: ''
```
Then the following ones have the correct frame_id.

I let the issue open for now, I will repeat the test to assess whether it is erratic or not.

closed 2023-09-27 by lv-alerion