rsasaki0109/lidar_slam_ros2 #79: Scan_Matcher "odom"-Frame Error

LuSchnitt, 2023-09-26:
Hello, 

the installation of the package went very smoothly and without problems.
But when i try to test it, with my ros2-bag, i got the following error only after i started to publish the data from the bag:

[image]

System: ubuntu 20.04
Ros: Foxy
Arm64 Architecture

I was working with the scanmatcher a few months ago and i didnt get any error like this with the tested ros_bags. 

Do you know how i can get solve this problem?

If you need more information just ask.

Best 

Lukas

rsasaki0109, 2023-09-26:
Is the scanmatcher.use_odom in your yaml set to true? If you don't have odometry as a sensor input, you need to set use_odom to false.

LuSchnitt, 2023-09-26:
Thanks, that solved the problem!

Best Lukas

closed 2023-09-26 by LuSchnitt