rsasaki0109/lidar_slam_ros2 #81: Help Needed with ROS2 LidarSLAM: QoS Policy Errors and Map Not Publishing

brian7989, 2023-12-01:
Hello,

I am currently working on a project using ROS2 with a TurtleBot3 in a Webots simulation environment, and I'm encountering issues with LidarSLAM. My primary problem is that my map isn't being published as expected. I suspect this might be related to QoS policy mismatches, but I'm not entirely sure. I would greatly appreciate any insights or suggestions you might have. Here are the relevant details and logs:

**Environment:**

ROS2 Humble
TurtleBot3 in a Webots simulation
LidarSLAM for mapping

**Issue:**

The map is not being published.
Receiving warnings about incompatible QoS policies, specifically related to DURABILITY_QOS_POLICY.

**Relevant Logs:**

```
[static_transform_publisher-4] [WARN] [1701458360.405101586] []: Old-style arguments are deprecated; see --help for new-style arguments
[robot_state_publisher-3] [WARN] [1701458360.419031169] [robot_state_publisher]: New subscription discovered on topic '/tf', requesting incompatible QoS. No messages will be sent to it. Last incompatible policy: DURABILITY_QOS_POLICY
[lidarslam-6] [WARN] [1701458366.354324463] [lidarslam]: New subscription discovered on topic '/tf', requesting incompatible QoS. No messages will be sent to it. Last incompatible policy: DURABILITY_QOS_POLICY
[webots_controller_TurtleBot3Burger-5] [WARN] [1701458366.348362380] [TurtleBot3Burger]: New subscription discovered on topic '/scan', requesting incompatible QoS. No messages will be sent to it. Last incompatible policy: DURABILITY_QOS_POLICY
```

**Attempts to Resolve:**

- Checked network configuration and ensured all nodes are on the same ROS2 network.
- Verified hardware/simulation settings.
- Attempted to align QoS settings across the system.

Despite these efforts, the issue persists. I'm unsure if the QoS policy mismatch is the sole reason for the map not publishing, or if there are other underlying issues. Any advice or guidance on how to troubleshoot and resolve this would be greatly appreciated.

Thank you in advance for your help!

rsasaki0109, 2023-12-02:
The 'scam_mathcer' node might not be performing scan matching. You can check if the input scan topic is properly fed into the 'scan_mathcer' node using 'ros2 topic info -v'. By setting the 'scan_mathcer debug_flag' to true in the YAML file, you can view more detailed logs.

https://github.com/rsasaki0109/lidarslam_ros2/blob/develop/lidarslam/param/lidarslam.yaml#L28

brian7989, 2023-12-04:
Fixed the issue by adding a correct remappings for scan_matcher

  ```
lidarslam_params = os.path.join(package_dir, 'resource', 'lidar_slam.yaml')
  scan_matcher = Node(
      package='scanmatcher',
      executable='scanmatcher_node',
      parameters=[lidarslam_params],
      remappings=[('/input_cloud','/scan/point_cloud')],
      output='screen',
      condition=launch.conditions.IfCondition(use_lidar_slam)
  )

  graph_based_slam = Node(
      package='graph_based_slam',
      executable='graph_based_slam_node',
      parameters=[lidarslam_params],
      output='screen',
      condition=launch.conditions.IfCondition(use_lidar_slam)
  )
```

Thank you so much for your help!

closed 2023-12-04 by brian7989