rsasaki0109/lidar_slam_ros2 #82: Use with Carla Bag

guzgonav, 2023-12-21:
Hi, I've been trying to make it work with Carla but I've not been able to. 

The sample bag (hdl_400) works just fine, I have a ros2 bag created with Carla that contains imu, lidar and tf amongst others. 
I changed the topic that receives the PointCloud2 from /velodyne_points to my carla topic, having this: 
**lidarslam.launch.py**
```
mapping = launch_ros.actions.Node(
        package='scanmatcher',
        executable='scanmatcher_node',
        parameters=[main_param_dir],
        remappings=[('/input_cloud','/carla/ego_vehicle/lidar')],
        output='screen'
        )
```
        
I tried to change the tf to: 
```
tf = launch_ros.actions.Node(
        package='tf2_ros',
        executable='static_transform_publisher',
        arguments=['0','0','0','0','0','0','1','ego_vehicle','ego_vehicle/lidar']
				)
```
As my tf_tree is: 
[frames_2023-12-20_23.13.49.pdf](https://github.com/rsasaki0109/lidarslam_ros2/files/13733826/frames_2023-12-20_23.13.49.pdf)

But it keeps saying: 
_[scanmatcher_node-1] [ERROR] [1703118618.801740320] [scan_matcher]: "base_link" passed to lookupTransform argument target_frame does not exist.

Any idea what is the error? 
(Im using ros2 humble)
Thank u in advance.

rsasaki0109, 2023-12-25:
It seems necessary to also change the parameters related to tf on the lidarslam_ros2 side.

>     robot_frame_id: "base_link"
https://github.com/rsasaki0109/lidarslam_ros2/blob/develop/lidarslam/param/lidarslam.yaml#L4

guzgonav, 2023-12-27:
Thanks, I found out what was causing the error, the correct paramers are: 
```
mapping = launch_ros.actions.Node(
        package='scanmatcher',
        executable='scanmatcher_node',
        parameters=[main_param_dir],
        remappings=[('/input_cloud','/carla/ego_vehicle/lidar')],
        output='screen'
        )

    tf = launch_ros.actions.Node(
        package='tf2_ros',
        executable='static_transform_publisher',
        arguments=['0','0','0','0','0','0','1','base_link','/ego_vehicle']
```

After that change it works, I also modified the parameter you pointed out, from: 
`robot_frame_id: "base_link"`
To: 
robot_frame_id: "ego_vehicle"
And it keep working fine.

rsasaki0109, 2023-12-28:
It seems this Issue has been resolved, so I'm closing it. Thank you for your report.

closed 2023-12-28 by rsasaki0109

wxyLuna2024, 2024-07-03:
Hi, 
Can you share the .yaml parameters? I am trying to run the kitti bag but is encountering ndt failing issues