rsasaki0109/lidar_slam_ros2 #83: About the use in the window environment

lidarmansiwon, 2024-02-13:
Hello, thanks to you, I used it so well. Aside from that, I'm trying to localize your slam and hdl in a window environment using ouster lidar. Is this possible? I have a lot of worries such as using tf and ndt. I'm asking because you are much better than me. Is it possible if I adjust the input and output values of lidar data well?
If you have any advice, please be polite.

rsasaki0109, 2024-02-20:
So, you're planning to use Windows instead of Ubuntu, right? I'm not very familiar with the details, but could it work by just installing ROS? It seems like there might be bigger issues beyond localization to consider. If you're stuck, I recommend looking into it further and, if you still have questions, asking on platforms like https://robotics.stackexchange.com/ might be a good idea.

lidarmansiwon, 2024-02-26:
Thank you for your kind reply. I feel limited in the window and I'm going to use docker. But I'm currently experiencing an error that doesn't connect to the ouster. Anyway, thank you for your kind reply!