rsasaki0109/lidar_slam_ros2 #84: Ouster OS1-16 lidar usage

Mustafa19077, 2024-03-21:
How can ı use Ouster Lidar with this repo?. ı can run with hdl_400 bag file. But when ı try to use ouster bag with topic /ouster/points the rviz2 doesnt show the data and ı get this error = "[rviz2-4] [INFO] [1710964446.672514711] [rviz]: Message Filter dropping message: frame 'os_lidar' at time 544.149 for reason 'discarding message because the queue is full'

also there is one more problem, ı think also need to change scanmatcher base_link with os lidar, 
how can ı solve this problem

closed 2024-03-21 by Mustafa19077

youweik2, 2024-08-08:
Could you tell me how to solve it? I met the same problem with you. Any help much will be appreciated.

CStubenvoll, 2024-10-28:
I have the same problem. If anyone solved it, could you post the solution? 
(I also set os_lidar as the base_link in the lidarslam.launch.py file (which might be a problem when running the lidarslam not on a bag but with the real sensor since os_lidar already has os_sensor as a parent there and this change sets base_link as the parent for os_lidar)).

Mustafa19077, 2024-10-28:
> I have the same problem. If anyone solved it, could you post the solution? (I also set os_lidar as the base_link in the lidarslam.launch.py file (which might be a problem when running the lidarslam [...]

Hi, I have solved this problem long time ago. and this is the message at those time. I hope it helps you. Let me know work or not.

Thanks for response. Because of the letter limit ı didn’t explain details. Now ı will explain. 
I have solved the problem for lidarslam_ros2. I changed the topic name /input_cloud “velodyne_points” to “ouster_points” and base_link “velodyne” to “os_lidar” in lidarslam.launch.py. Also ı change robot_frame_id: "base_link" to “os_lidar” inside /param/lidarslam.yaml. After this change, ı can use this repo and Everything is fine. Thanks again for your project. 

Now ı have another problem. When ı run lidar_localization_ros2 ı get some error. I make the some changes as ı said above, but unfortunately I didn’t. But I found a way that i use this repo. Here is the solution
I realize the algorithm use topic name as “velodyne_points” and frame id as “velodyne”. I create a subscriber (subscribe to ouster/points) and after get the msg:PointCloud2 , ı change msg.header._frame_id = “velodyne”. Then ı create a publisher that use topic name “velodyne_points”, I publish this new PointCloud2 message with this publisher, everything is fine. With this 2 step, I can create my map with lidarslam_ros2 and make localization. 

But there must be solution for that. I don’t think it is good solution for localization. Now ı want to ask what kind of changes must be made when ı use different lidar that use different topic_name and frame_id. I didn’t find a documentation for that, maybe this is my faul. Because ı am new with ROS2.