rsasaki0109/lidar_slam_ros2 #85: plz help me, ros-foxy- libg2o

LEELEE412, 2024-04-01:
I've been suffering from this issue for a few days. I've installed all the files. I got the foxy version. After installing, when I do the rosdep installation, I get the following error.

$rosdep install --from-paths src --ignore-src -r -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
scanmatcher: Cannot locate rosdep definition for [libg2o]
graph_based_slam: Cannot locate rosdep definition for [libg2o]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully

If I ignore and proceed with the installation

Processing: ndt_omp_ros2]                                       
--- stderr: ndt_omp_ros2                                          
In file included from /usr/local/include/pcl-1.14/pcl/pcl_base.h:47,
                 from /usr/local/include/pcl-1.14/pcl/filters/filter.h:42,
                 from /usr/local/include/pcl-1.14/pcl/filters/voxel_grid.h:42,
                 from /home/sejin/slam_ws/src/lidarslam_ros2/Thirdparty/ndt_omp_ros2/include/pclomp/voxel_grid_covariance_omp.h:42,
                 from /home/sejin/slam_ws/src/lidarslam_ros2/Thirdparty/ndt_omp_ros2/src/pclomp/voxel_grid_covariance_omp.cpp:1:
/usr/local/include/pcl-1.14/pcl/filters/boost.h:46:1: note: #pragma message: This header is deprecated. Please include the needed boost headers directly. (It will be removed in PCL 1.15)
   46 | PCL_DEPRECATED_HEADER(1, 15, "Please include the needed boost headers directly.")
      | ^~~~~~~~~~~~~~~~~~~~~
/usr/local/include/pcl-1.14/pcl/filters/boost.h:46:1: note: #pragma message: This header is deprecated. Please include the needed boost headers directly. (It will be removed in PCL 1.15)
/usr/local/include/pcl-1.14/pcl/filters/boost.h:46:1: note: #pragma message: This header is deprecated. Please include the needed boost headers directly. (It will be removed in PCL 1.15)
/usr/local/include/pcl-1.14/pcl/filters/boost.h:46:1: note: #pragma message: This header is deprecated. Please include the needed boost headers directly. (It will be removed in PCL 1.15)
---
Finished <<< ndt_omp_ros2 [3min 3s]
Starting >>> graph_based_slam
Starting >>> scanmatcher                            
[Processing: graph_based_slam, scanmatcher]                                                         
--- stderr: scanmatcher

there is an error like this.

rsasaki0109, 2024-04-04:
It seems the PCL 1.14 you installed on your own is causing trouble.

rsasaki0109, 2024-04-04:
I think g2o can be installed with 'sudo apt install ros-foxy-libg2o'.

LEELEE412, 2024-04-09:
> I think g2o can be installed with 'sudo apt install ros-foxy-libg2o'.

I already installed this.

LEELEE412, 2024-04-09:
> It seems the PCL 1.14 you installed on your own is causing trouble.

okay i will delete pcl 1.14 and try again. 

My question is if veodyne is SLAM working on foxy, is this possible?

rsasaki0109, 2024-04-09:
Using this package, you can perform SLAM with Velodyne on ROS2 Foxy.

rsasaki0109, 2024-06-29:
Due to inactivity, I'm closing this issue. If there are still concerns, please open a new issue. Thank you.

closed 2024-06-29 by rsasaki0109