rsasaki0109/lidar_slam_ros2 #88: General Guidance

rvxfahim, 2024-04-29:
I have the understanding that to perform 2D localization, the workflow is to first scan the environment while slam toolbox is running and at one point it has to be saved as a map file which can then be loaded later to perform localization.
I am now trying to first generate the map but using 3D lidar from Gazebo velodyne simulator, but I understand that for generating the map I would also need IMU so where would that come from in the Gazebo simulation? 
I have built a 3d model of a parking lot as a obj file which I plan to scan through gazebo, and once it is done then save it as a map file and later load it again to perform the localization.

Hence I am looking for some guidance on what would be the workflow and required packages. 
what if I already have a pointcloud file e57 (leica blk360) which can also be converted to a laz file and viewed in cloudconvert. Then how can i use it to perform localization only?

rsasaki0109, 2024-05-04:
> for generating the map I would also need IMU 

IMUs are not essential for map generation.

> where would that come from in the Gazebo simulation?

I don't use Gazebo much, so I don't really remember where the IMU would come from in Gazebo simulations.

> Then how can i use it to perform localization only?

If you're using ROS, most ROS localization packages typically utilize the PCL library. You might be able to use your e57 file by converting it into the PCD format, which is the standard point cloud format used by PCL.

rsasaki0109, 2024-06-29:
Due to inactivity, I'm closing this issue. If there are still concerns, please open a new issue. Thank you.

closed 2024-06-29 by rsasaki0109