rsasaki0109/lidar_slam_ros2 #92: Frame Changes

theinge9, 2024-06-12:
I am trying this repository with my 3D LiDAR and it perfectly works in simple environment which has no many direction changes.
But when i try in the environment with a lot of corner, my LiDAR frame changes a lot and cannot find loop closures. How should i fix this in which config parameters?Thanks for your great work. 
I am using only 3D LiDAR without IMU.

[image]

rsasaki0109, 2024-06-17:
It seems like the scan matching is failing around the bottom right of the image.
Could you provide the parameter YAML file you are using?

theinge9, 2024-06-18:
I am using the following yaml file which is not different with the origin of lidarslam.
```
scan_matcher:
  ros__parameters:
    global_frame_id: "map"
    robot_frame_id: "base_link"
    registration_method: "NDT"
    ndt_resolution: 2.0
    ndt_num_threads: 2
    gicp_corr_dist_threshold: 5.0
    trans_for_mapupdate: 1.5
    vg_size_for_input: 0.5
    vg_size_for_map: 0.1
    use_min_max_filter: true
    scan_min_range: 1.0
    scan_max_range: 200.0
    scan_period: 0.2
    map_publish_period: 15.0
    num_targeted_cloud: 20
    set_initial_pose: true
    initial_pose_x: 0.0
    initial_pose_y: 0.0
    initial_pose_z: 0.0
    initial_pose_qx: 0.0
    initial_pose_qy: 0.0
    initial_pose_qz: 0.0
    initial_pose_qw: 1.0
    use_imu: false
    use_odom: false
    debug_flag: false

graph_based_slam:
    ros__parameters:
      registration_method: "NDT"
      ndt_resolution: 1.0
      ndt_num_threads: 2
      voxel_leaf_size: 0.1
      loop_detection_period: 3000
      threshold_loop_closure_score: 0.7
      distance_loop_closure: 100.0
      range_of_searching_loop_closure: 20.0
      search_submap_num: 2
      num_adjacent_pose_cnstraints: 5
      use_save_map_in_loop: true
      debug_flag: true
```

rsasaki0109, 2024-06-18:
Since the original file is designed for outdoor use, try adjusting the ndt_resolution to 1.0, vg_size_for_input to between 0.1 and 0.5, and trans_for_mapupdate to 1.0. It seems better to increase the ndt_num_threads as much as possible because it's too small.

theinge9, 2024-06-20:
I got this result with ndt_resolution to 1.0, vg_size_for_input to 3.0,  trans_for_mapupdate to 1.0 and ndt_num_threads to 15. Although the map is getting better, it still needs to adjust the level of frame. How should i control the frame level changes.?It may be due to the effect of my data.
[image]

Always thanks for your support.

rsasaki0109, 2024-06-20:
It might be good to halve ndt_resolution, vg_size_for_input, and vg_size_for_map. If the processing cannot keep up, please play the rosbag slower. Is the scan matching failing at the door area or when it rotates at the narrow staircase? You may need to tune the parameters to avoid misalignment there. In the worst case, you might need to remeasure that part slowly

theinge9, 2024-08-07:
Dear rsasaki,

Although really feel sorry for asking so many questions, may I know the difference between using IMU in this package and li_slam_ros2 which is a combination of LIO-SAM IMU composite method.

rsasaki0109, 2024-08-08:
This package's IMU only corrects for scan distortion and does not use the timestamp of each point in the point cloud, which slightly reduces accuracy. This is because it is based on the older lego-loam rather than lio-sam.
li_slam_ros2 utilizes both the IMU and the timestamps of each point in the point cloud for distortion correction and initial position estimation in scan matching, resulting in better performance