rsasaki0109/lidar_slam_ros2 #95: Please help noob

IIIIllllIllI, 2024-07-25:
I'm ros2 noob.
I use ouster lidar for slam.
I want to know what I have to change to use this package.

rsasaki0109, 2024-07-26:
The velodyne_points in the mapping should work if you change them to the ouster point topic and also change the frame ID of velodyne in tf to the frame ID of the ouster point topic.

https://github.com/rsasaki0109/lidarslam_ros2/blob/develop/lidarslam/launch/lidarslam.launch.py#L24-L36

IIIIllllIllI, 2024-08-20:
Thank you for your answer. Now, I can use your package. However, I have two more questions. 

1. When I launched this package and played the bag file, I couldn't see the /modified_path and /modified_map. When I echoed the topics, no information came out. I think there might be an issue with the graph-based-slam node.

2. I didn't change any parameters without ndt_num_threads. I changed ndt_num_threads to 4. While it works well in simple environments, it doesn't perform as expected in more complex environments. How can I adjust the parameters in this situation. 

file:///home/user/Pictures/Screenshots/Screenshot%20from%202024-08-21%2002-56-53.png

file:///home/user/Pictures/Screenshots/Screenshot%20from%202024-08-21%2003-05-57.png

IIIIllllIllI, 2024-08-20:
[image]

IIIIllllIllI, 2024-08-20:
[image]

rsasaki0109, 2024-08-21:
Graph-based SLAM will not publish trajectories and maps unless you perform a loop closure or execute "ros2 service call /map_save std_srvs/Empty". If you do not perform a loop closure, please execute "ros2 service call /map_save std_srvs/Empty".

The point cloud map in the image does not seem to have any obvious problems, but what is the issue?

Generally, you need to adjust parameters such as ndt_resolution and vg_size_for_input depending on whether the environment is indoors or outdoors.

IIIIllllIllI, 2024-08-21:
Thanks, I understand the problem.