rsasaki0109/lidar_slam_ros2 #96: General Advice for running LidarSlam and the IMU

Tom-Hickling, 2024-08-15:
We are using your package for odometry for a drone in an indoor environment. Is there any advice on settings for the package to get the best out of it in this sort of environment? We are using the IMU addition, and we were wondering what the scan_time needs to be changed to, as you mentioned changing it in the read-me when using the IMU. We were also wondering if you need to do the transformation between the IMU location and the lidar.

rsasaki0109, 2024-08-21:
Is 'scan_time' the same as 'scan_period'? Please set the 'scan_period' to the time interval of the LiDAR scan. For example, if it’s 10Hz, then the 'scan_period' should be 0.1 sec. The IMU values need to be transformed to match the LiDAR origin. This package doesn’t implement that, but you can use functions like the one in the link below:
https://github.com/rsasaki0109/transfrom_velocity/blob/main/src/transfrom_IMU.cpp
If you plan to use an IMU, you might want to consider using 'li_slam_ros2' instead of this package."

Tom-Hickling, 2024-08-23:
Thanks for the info. We're having issues maintaining an accurate z-position estimate. We think this is due to the enclosed nature of our environment. The mapping seems to work fine. There is just a problem when using the lidarslam for the odometry. Do you have any insight into this?

rsasaki0109, 2024-08-24:
Integrating the IMU can help reduce drift in the z-direction