rsasaki0109/lidar_slam_ros2 #98: I am using Unitree Go2, but since I am a beginner, I don't fully understand everything.

umsh315, 2024-10-07:
Hello Sasaki-san,

I am trying to use this package with the 4D LiDAR L1 mounted on the head of the Unitree Go2 quadruped robot. I modified the launch file by changing remappings=[('/input_cloud', '/velodyne_points')] to /utlidar/cloud_deskewed and arguments=['0', '0', '0', '0', '0', '0', '1', 'base_link', 'velodyne'] to odom, but it didn't work well. The path appears in a spiral shape, with irregular point clouds scattered around its center, and I am not sure what is happening. If you have any insights, I would greatly appreciate your assistance. Thank you in advance.

rsasaki0109, 2024-10-08:
I am curious about the current state of the RViz display. Additionally, if you are working indoors, it might be helpful to adjust the parameters as described below:

https://github.com/rsasaki0109/lidarslam_ros2/issues/28#issuecomment-840052272

XX-4869, 2025-02-27:
> 您好 Sasaki-san，
> 
> 我正在尝试将此软件包与安装在 Unitree Go2 四足机器人头部的 4D LiDAR L1 一起使用。我通过将 remappings=[（'/input_cloud'， '/velodyne_points'）] 更改为 /utlidar/cloud_deskewed并将 arguments=['0'， '0'， '0'， '0'， '0'， '1'， 'base_link'， 'velodyne'] 更改为 odom 来修改启动文件，但它效果不佳。路径呈螺旋状，不规则的点云散布在其中心周围，我不确定发生了什么。如果您有任何见解，我将非常感谢您的帮助。提前谢谢你。

Hello, I am also using go2 at present. I would like to discuss it with you. Could you give me the contact information