rsasaki0109/lidar_slam_ros2 #101: Quest to loop closing

elorenz97, 2024-11-08:
Hello,
First of all, thank you for creating this lidar-based SLAM. In particular, that it works for ROS2 Humble. :) 
So far, it's working really well. I use it with a Velodyne VLP-16 for outdoor use. 

I want to show the differences between with and without loop closure, as you did in an example. But I'm a beginner with loop closing. So I've already noticed that the normal path is without loop closure and the modified path is with loop closing. 
However, on the picture you can see that it looks like the modified path has a lower "resolution" than the yellow path. Especially in the curve it looks really bad. 
But I don't know why? What parameter can I change to fix the problem?

[image]

rsasaki0109, 2024-11-08:
The yellow path represents the LiDAR odometry path, and its frequency matches the LiDAR’s cycle.

The green path is a collection of poses used in graph optimization, which are downsampled compared to the yellow path. Loop closure is achieved by dividing the map into submaps, performing matching between submaps to search for loops, and the green path represents a collection of representative poses of these submaps.

Therefore, the lower resolution of the green path is by design. To increase the resolution, code modifications would be necessary.

elorenz97, 2024-11-11:
oh I understand.
Thank you for your explanation. :)

elorenz97, 2024-11-11:
But how did you achived this result?
https://github.com/rsasaki0109/lidarslam_ros2/blob/develop/lidarslam/images/path.png
Can I disabled loopclosure?