rsasaki0109/lidar_slam_ros2 #105: Issue with mapping indoors

willgpaik, 2025-01-01:
Hello,
I'm trying to use your package to map indoor areas (not sure it's suitable for small area though). I'm testing with 2 different 3D lidars, Aeva Aeris2 and Intel Realsense L515. For Aeva lidar, I'm using following configurations:
[image]
and
[image]
I'm having trouble mapping my lab area due to few issues. With fixed frame, aeva/vehicle, the frame keeps rotating and the map doesn't get stitched even when I'm simply rotating the lidar around 180 degrees. Also, for some reason, I'm getting a strange path while rotating the lidar:
[image]
I'm not sure what's causing the issue and how to fix it.
For Realsense L515, I was trying to map my own room with similar configuration but getting some useless map:
[image]
Could you give me some suggestions to get better results?

rsasaki0109, 2025-01-27:
Sorry for the late reply. It's difficult to perform scan matching without a rotating LiDAR... It's especially sensitive to rotation, so please rotate it slowly. Instead of adjusting the parameters, it might be better to use something like the Livox Mid-360.

FabianJohnThl, 2025-02-26:
We`re trying to reproduce the indoor scans with a MID360 and get the same bad results with drifting/incorrect path or pose. Is there any example configuration for the MID360 that is working?

rsasaki0109, 2025-02-27:
I feel like it might work with the MID360 if IMU is not used, but I have never tested it. However, the IMU in this package does not work with the MID360.

FabianJohnThl, 2025-02-27:
Thanks for your fast response. It seems, that the package works better without the IMU.
But after moving the lidar slightly, the created map  looks like this (the position seems to be rotated):

[image]

Any guess how to fix this? Or maybe, can you explain, why the Livox IMU of the MID360 isn't working with the package? Is it caused by the message format of the IMU? What is the expected format, than I could write a wrapper for this.
Best regards