rsasaki0109/lidar_slam_ros2 #108: some problems

XX-4869, 2025-02-20:
hi！
when I run 
rosdep install --from-paths src --ignore-src -r -y

the terminal says 
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
scanmatcher: Cannot locate rosdep definition for [libg2o]
graph_based_slam: Cannot locate rosdep definition for [libg2o]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully

I do not know how to handle this problem.Can you give me some advice?

rsasaki0109, 2025-02-21:
Is the ROS distro Jazzy? I'm not very familiar with Jazzy, but there might not be a binary package for g2o.

Just in case, please try the following:

```
sudo apt update
sudo apt install ros-${ROS_DISTRO}-g2o
```
If that doesn't work, you'll need to install it from the source:

```
cd ~
git clone https://github.com/RainerKuemmerle/g2o.git
cd g2o
mkdir build && cd build
cmake ..
make -j$(nproc)
sudo make install
```

XX-4869, 2025-02-22:
hi，
I use ros-foxy 
when I run
sudo apt install ros-foxy-g2o
It says:
正在读取软件包列表... 完成
正在分析软件包的依赖关系树       
正在读取状态信息... 完成       
E: 无法定位软件包 ros-foxy-g2o

when I use git to install g2o, it still have some errors.
unitree@ubuntu:~/lidarslam_ros2_ws$ rosdep install --from-paths src --ignore-src -r -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
scanmatcher: Cannot locate rosdep definition for [libg2o]
lidarslam: Cannot locate rosdep definition for [pcl_conversions]
graph_based_slam: Cannot locate rosdep definition for [libg2o]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully

XX-4869, 2025-02-22:
> Is the ROS distro Jazzy? I'm not very familiar with Jazzy, but there might not be a binary package for g2o.
>
> Just in case, please try the following:
>
> ```
> sudo apt update
> sudo apt install [...]

hi，
I use ros-foxy
when I run
sudo apt install ros-foxy-g2o
It says:
正在读取软件包列表... 完成
正在分析软件包的依赖关系树
正在读取状态信息... 完成
E: 无法定位软件包 ros-foxy-g2o

when I use git to install g2o, it still have some errors.
unitree@ubuntu:~/lidarslam_ros2_ws$ rosdep install --from-paths src --ignore-src -r -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
scanmatcher: Cannot locate rosdep definition for [libg2o]
lidarslam: Cannot locate rosdep definition for [pcl_conversions]
graph_based_slam: Cannot locate rosdep definition for [libg2o]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully

rsasaki0109, 2025-02-25:
Then, maybe the source code needs to be built.

XX-4869, 2025-02-26:
> 然后，可能需要构建源代码。

What should I do, please? I am a beginner. I hope you can teach me.

rsasaki0109, 2025-02-26:
You should be able to simply clone pcl_conversions from GitHub and place it under the src directory of your workspace.

You can install g2o using the following command. Once that’s done, run rosdep and then build with colcon build.
```
cd ~
git clone https://github.com/RainerKuemmerle/g2o.git
cd g2o
mkdir build && cd build
cmake ..
make -j$(nproc)
sudo make install
```

XX-4869, 2025-02-27:
> You should be able to simply clone pcl_conversions from GitHub and place it under the src directory of your workspace.
> 
> You can install g2o using the following command. Once that’s done, skip rosdep and then build with colcon build.
> 
> ```
> cd ~
> git clone https://github.com/RainerKuemmerle/g2o.git
> cd g2o
> mkdir build && cd build
> cmake ..
> make -j$(nproc)
> sudo make install
> ```

then,I have some problems that follows.
unitree@ubuntu:~/lidarslam_ros2_ws$ colcon build --symlink-install   --cmake-args   -DCMAKE_BUILD_TYPE=Release   -DPCL_DIR=/opt/pcl-1.12/share/pcl-1.12
Starting >>> lidarslam_msgs
Starting >>> ndt_omp_ros2
Finished <<< ndt_omp_ros2 [0.39s]                                      
Finished <<< lidarslam_msgs [0.67s]                  
Starting >>> graph_based_slam
Starting >>> scanmatcher
--- stderr: graph_based_slam                                             
In file included from /home/unitree/lidarslam_ros2_ws/src/lidarslam_ros2/graph_based_slam/src/graph_based_slam_component.cpp:1:
/home/unitree/lidarslam_ros2_ws/src/lidarslam_ros2/graph_based_slam/include/graph_based_slam/graph_based_slam_component.h:47:10: fatal error: tf2_sensor_msgs/tf2_sensor_msgs.hpp: 没有那个文件或目录
   47 | #include <tf2_sensor_msgs/tf2_sensor_msgs.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/graph_based_slam_component.dir/build.make:63：CMakeFiles/graph_based_slam_component.dir/src/graph_based_slam_component.cpp.o] 错误 1
make[1]: *** [CMakeFiles/Makefile2:107：CMakeFiles/graph_based_slam_component.dir/all] 错误 2
make: *** [Makefile:141：all] 错误 2
---
Failed   <<< graph_based_slam [0.50s, exited with code 2]
--- stderr: scanmatcher
In file included from /home/unitree/lidarslam_ros2_ws/src/lidarslam_ros2/scanmatcher/src/scanmatcher_component.cpp:1:
/home/unitree/lidarslam_ros2_ws/src/lidarslam_ros2/scanmatcher/include/scanmatcher/scanmatcher_component.h:47:10: fatal error: tf2_sensor_msgs/tf2_sensor_msgs.hpp: 没有那个文件或目录
   47 | #include <tf2_sensor_msgs/tf2_sensor_msgs.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/scanmatcher_component.dir/build.make:63：CMakeFiles/scanmatcher_component.dir/src/scanmatcher_component.cpp.o] 错误 1
make[1]: *** [CMakeFiles/Makefile2:80：CMakeFiles/scanmatcher_component.dir/all] 错误 2
make: *** [Makefile:141：all] 错误 2
---
Failed   <<< scanmatcher [0.51s, exited with code 2]

Summary: 2 packages finished [1.72s]
  2 packages failed: graph_based_slam scanmatcher
  2 packages had stderr output: graph_based_slam scanmatcher
  1 package not processed

rsasaki0109, 2025-02-27:
First, check if tf2_sensor_msgs is installed on your system by running the following command:

```
ros2 pkg list | grep tf2_sensor_msgs
```
If tf2_sensor_msgs is not displayed in the output, it means that it is not installed.
In that case, install it using the following command:

```
sudo apt install ros-foxy-tf2-sensor-msgs
```

XX-4869, 2025-02-27:
> 首先，通过运行以下命令检查您的系统上是否安装了 tf2_sensor_msgs：
> 
> ```
> ros2 pkg list | grep tf2_sensor_msgs
> ```
> 
> 如果输出中未显示 tf2_sensor_msgs，则表示未安装该 ID。在这种情况下，请使用以下命令安装它：
> 
> ```
> sudo apt install ros-foxy-tf2-sensor-msgs
> ```

I did, but in ros-foxy, it is.h instead of .hpp. Can you compile it by changing it to .h

rsasaki0109, 2025-02-27:
You're right

XX-4869, 2025-02-27:
if i want to use the lidar L1 on unitree go2 , how to modify the project
this is the topic of pointcloud:/utlidar/cloud
or /utlidar/cloud_deskewed

rsasaki0109, 2025-02-27:
I haven't used LiDAR L1 before, but I think so. You need to modify the launch file.

XX-4869, 2025-02-27:
> I haven't used LiDAR L1 before, but I think so. You need to modify the launch file.

how to modify?where should I  change the pointcloud topic?

rsasaki0109, 2025-02-27:
https://github.com/rsasaki0109/lidarslam_ros2/blob/develop/lidarslam/launch/lidarslam.launch.py