rsasaki0109/lidar_slam_ros2 #109: Installing on ROS2 Humble

anirudh-kilambi, 2025-02-21:
Hello, when installing the package on a Clearpath Husky running ROS2 humble, we faced an issue when running the colcon build. Below is the error message:
```

Starting >>> lidarslam_msgs
Finished <<< lidarslam_msgs [6.05s]                     
Starting >>> graph_based_slam
Starting >>> scanmatcher
--- stderr: scanmatcher                                                                     
CMake Error at CMakeLists.txt:33 (find_package):
  By not providing "Findpcl_conversions.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "pcl_conversions", but CMake did not find one.

  Could not find a package configuration file provided by "pcl_conversions"
  with any of the following names:

    pcl_conversionsConfig.cmake
    pcl_conversions-config.cmake

  Add the installation prefix of "pcl_conversions" to CMAKE_PREFIX_PATH or
  set "pcl_conversions_DIR" to a directory containing one of the above files.
  If "pcl_conversions" provides a separate development package or SDK, be
  sure it has been installed.

---
Failed   <<< scanmatcher [1.37s, exited with code 1]
Aborted  <<< graph_based_slam [1.38s]

Summary: 1 package finished [7.76s]
  1 package failed: scanmatcher
  1 package aborted: graph_based_slam
  2 packages had stderr output: graph_based_slam scanmatcher
  1 package not processed
```

I noticed when trying to build pcl_conversions from source on github that the repo had been archived. Is there any preferred way of dealing with this package dependency? 

Thanks!

rsasaki0109, 2025-02-21:
The options for addressing this issue are either building the source code of pcl_conversions or modifying the lidarslam_ros2 code to align with the fact that pcl_conversions has likely been moved to another package.

anirudh-kilambi, 2025-02-28:
Thanks for your input! We were able to figure it out by reading the actual error messages and updating the CMakeLists as required. Closing!

closed 2025-02-28 by anirudh-kilambi