rsasaki0109/lidar_slam_ros2 #111: Setting up lidarslam_ros2 for other LiDARs

DevankGarg, 2025-02-27:
Hello,

I am trying to create a PCD file of an open environment. I am using Carla simulator to get the LiDAR data. Can you please describe where should I make changes in the code files to adapt it for my LiDAR sensor configuration?

Thanks in advance.

rsasaki0109, 2025-02-27:
I think just changing the launch and YAML files should be enough. The launch file is especially important.

DevankGarg, 2025-02-27:
[image]
Hi, Thanks for replying. I have made these changes in launch file 'lidarslam.launch.py'. Do I need to change in lidarslam_tukuba.launch.py as well? And in which YAML file should I change?

rsasaki0109, 2025-03-13:
Sorry for the late reply. You choose which one to launch at startup, so please modify the launch file for the one you select. If you’re unsure, I recommend going through the ROS tutorial first…