rsasaki0109/lidar_slam_ros2 #112: Enabling Odometry

anirudh-kilambi, 2025-03-13:
Hi again, 

Running LiDAR SLAM has no problems now. We'd like to enable odometry. Inside `~/ros2_ws/src/lidarslam_ros2/lidarslam/param/lidarslam.yaml` I've updated it such that:

```
scan_matcher:
  ros__parameters:
    use_odom: true
```

When launching lidarslam, I get the following error:

`[scan_matcher]: "odom" passed to lookupTransform argument target_frame does not exist`

We had a similar issue initially with our Velodyne VLP-16. We had to update the `mapping` variable in `lidarslam.launch.py`

```
remappings = [('/input_cloud', '/husky3/sensors/lidar3d_0/points')]
```

We also updated the `tf` variable:

```
arguments=['0', '0', '0', '0', '0', '0', '1', 'base_link', 'lidar3d_0_laser']
```

Our odom topic is `/husky3/platform/odom`. I'm wondering if there is some remap that needs to be done. One thing to note is that if I run `ros2 run tf2_tools view_frames`, there is no `odom` frame at all. However if I `ros2 topic echo /husky3/platform/odom --once`, I can see that `frame_id: odom` and `child_frame_id: base_link`

Any ideas what might need to be updated? I'm unsure if the tf or the remap variable needs to be updated, or if I'm missing something. Any help would be appreciated!

rsasaki0109, 2025-03-13:
It is necessary not only to publish the odom topic but also to send the TF.

anirudh-kilambi, 2025-03-13:
Thank you! That makes sense. Do you have any tips on synchronizing the clock times? Currently the odom publishes at 0.02 seconds slower than the velodyne's /points topic.

rsasaki0109, 2025-03-13:
Hmm, I haven’t had any issues with that, so I don’t have any tips. Could it be that the package publishing odom isn’t working well?

anirudh-kilambi, 2025-03-14:
I wrote my own publisher to publish the odometry to /tf, looks something like this:

```
import rclpy
from rclpy.node import Node
from nav_msgs.msg import Odometry
from tf2_ros import TransformBroadcaster
from geometry_msgs.msg import TransformStamped

class OdomTFBroadcaster(Node):
    def __init__(self):
        super().__init__('odom_tf_broadcaster')

        # Create a TransformBroadcaster
        self.tf_broadcaster = TransformBroadcaster(self)

        # Subscribe to odometry topic
        self.subscription = self.create_subscription(
            Odometry,
            '/husky3/platform/odom',  # Your odometry topic
            self.odom_callback,
            10)

    def odom_callback(self, msg):
        # Create a TransformStamped message
        t = TransformStamped()

        # Copy header information
        t.header = msg.header

        # Set child frame
        t.child_frame_id = msg.child_frame_id

        # Copy transform data from odometry
        t.transform.translation.x = msg.pose.pose.position.x
        t.transform.translation.y = msg.pose.pose.position.y
        t.transform.translation.z = msg.pose.pose.position.z
        t.transform.rotation = msg.pose.pose.orientation

        # Broadcast the transform
        self.tf_broadcaster.sendTransform(t)

def main(args=None):
    rclpy.init(args=args)
    node = OdomTFBroadcaster()
    rclpy.spin(node)
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```

Odom data is around 0.01 seconds behind, any tips?

rsasaki0109, 2025-03-18:
Hmm... Maybe implementing it in C++ instead of Python could improve it... maybe...

vidneu, 2025-06-10:
Did you find a (simple) workaround for this? Currently facing the same issue with a delay 0.01